hardbyte/python-can #809: cyclic task data changing

jma1123, 2020-04-15:
In trying the Broadcast Manager to send periodic messages with modified data from the 'read the docs', this does not seem very useful.  Ref: section 3.6 page 27&27

Rather a thread safe structure / object holding the data (say a python deque) or pointer to the data should be passed with the task creation.  The present implementation modifies the data 1x and sends this same data until the task is stopped.  Can you setup the implementation to send periodically, but the data is changing constantly, not just on the task instantiation?  Suggestions?  This limitation does not seem to be the intent of the periodic message task.

jma1123, 2020-04-21:
No one uses the periodic messages?  If this true then please ignore this item and indicate as much (I'll delete it).  The concept of the periodic message handler is awesome and would be wonderful feature, but it's not at all useful in the present state unless you are troubleshooting your CAN bus.

bessman, 2020-04-22:
This is a discussion that seems to come up every now and then, see for example https://github.com/hardbyte/python-can/pull/703. The issue is that some interfaces offer hardware support for message modification, while others don't. At the moment, there is no fallback implementation in software for those interfaces, and no consensus on how such an implementation would look.

You might also want to take a look at this example, which shows how to send a periodic message with a counter and checksum: https://github.com/hardbyte/python-can/blob/develop/examples/crc.py 

If the message(s) you want to send are predictable, a similar solution should work.

jma1123, 2020-04-22:
bessman - thank you for the reply; In my particular application the data is not predictable and rapidly changing.  If this is ever revisited I would welcome a discussion.  cheers

closed 2020-04-22 by jma1123