hardbyte/python-can #810: Issue with BLFReader

msnir009, 2020-04-16:
Hi, I am trying to read a blf file on a linux machine and i have done the following

import can
can.BLFReader("sample.blf")

error message
AttributeError: module 'can' has no attribute 'BLFReader'

I am not sure what am I doing wrong here as I have already installed the python-can package and it is still not recognising this method.

christiansandberg, 2020-04-17:
Which Python version and python-can version are you using? You don't happen to have a file named can.py in your folder?

msnir009, 2020-04-17:
it seems to have been an issue with the python version i was using which was python3. when I used python only, it worked,

karlding, 2020-05-22:
Closing this out as there doesn't seem to be anything actionable here. Please reopen if necessary.

closed 2020-05-22 by karlding