hardbyte/python-can #816: ASCReader can not read out CAN-FD frames from .ASC file

hdnauth, 2020-04-20:
Hi,
I have some trouble on loading CAN-FD log files (both .asc and .blf)
when I read .asc file using ASCRead(), only normal CAN frame is loaded.

1) How can I read CAN-FD frames?
2) How can I select one channel when loading?

Thanks.

<test input>
`date Thu Apr 16 01:59:01.000 pm 2020
base hex  timestamps absolute
internal events logged
// version 11.0.0
Begin TriggerBlock Thu Apr 16 01:59:01.000 pm 2020
   0.000000 Start of measurement
   0.001036 CANFD   1 Rx        29a  AAA_01                     1 1 8  8 f8 62 01 00 00 40 00 00   103625  136   307000 98005364 4ba00150 4b280150 20002776 2000091c
   0.001748 2  562             Rx   d 8 62 01 00 00 00 00 FF FF  Length = 233925 BitCount = 121 ID = 1378
   0.004038 2  562             Rx   d 8 62 04 00 00 00 00 FF FF  Length = 233925 BitCount = 121 ID = 1378
   0.009148 2  660             Rx   d 8 68 98 A9 02 50 14 00 00  Length = 227910 BitCount = 118 ID = 1632
   0.010804 CANFD   1 Tx        2e0  BBB_03                     0 0 c 24 00 00 00 00 45 50 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   584296  295   301040 b0141a48 4ba00150 4b280150 20002776 2000091c
   0.011538 CANFD   1 Tx        1a0  CCC_20                 0 0 d 32 00 00 00 fe f7 7f ff 00 14 00 10 00 10 00 02 00 00 00 00 00 00 00 00 00 fe 07 00 00 00 00 00 00   728312  367   321040 b008c4ec 4ba00150 4b280150 20002776 2000091c
   0.014092 2  661             Rx   d 8 02 00 00 00 00 00 00 00  Length = 239910 BitCount = 124 ID = 1633
   0.019098 2  662             Rx   d 8 41 30 31 32 31 31 46 00  Length = 219910 BitCount = 114 ID = 1634
   0.020624 CANFD   1 Tx        12a  CCC_30                 0 0 a 16 00 00 00 50 80 00 00 00 00 00 00 00 00 00 00 00   420312  213   301040 e0017971 4ba00150 4b280150 20002776 2000091c
   0.020901 CANFD   1 Tx        345  CCC_01                0 0 8  8 00 00 00 04 00 00 00 00   270312  138   321040 b0004e0b 4ba00150 4b280150 20002776 2000091c
   0.021635 CANFD   1 Tx        1a0  CCC_20                 0 0 d 32 00 00 00 fe f7 7f ff 00 14 00 10 00 10 00 02 00 00 00 00 00 00 00 00 00 fe 07 00 00 00 00 00 00   728312  367   321040 b008c4ec 4ba00150 4b280150 20002776 2000091c
   0.022225 CANFD   1 Tx        2e0  BBB_03                     0 0 c 24 00 00 00 00 45 50 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   584296  295   321040 b0141a48 4ba00150 4b280150 20002776 2000091c
   0.200692 2  4CE             Tx   d 8 10 C2 FF FF 00 00 80 00  Length = 238259 BitCount = 122 ID = 1230
End TriggerBlock
`

<test output>
`Timestamp:        0.001748        ID: 0562    S                DLC:  8    62 01 00 00 00 00 ff ff     Channel: 1
Timestamp:        0.004038        ID: 0562    S                DLC:  8    62 04 00 00 00 00 ff ff     Channel: 1
Timestamp:        0.009148        ID: 0660    S                DLC:  8    68 98 a9 02 50 14 00 00     Channel: 1
Timestamp:        0.014092        ID: 0661    S                DLC:  8    02 00 00 00 00 00 00 00     Channel: 1
Timestamp:        0.019098        ID: 0662    S                DLC:  8    41 30 31 32 31 31 46 00     Channel: 1
Timestamp:        0.200692        ID: 04ce    S                DLC:  8    10 c2 ff ff 00 00 80 00     Channel: 1`

[logtest.zip](https://github.com/hardbyte/python-can/files/4502490/logtest.zip)

karlding, 2020-04-22:
I believe being able to read CAN FD frames is addressed in `develop` by #741 in c92fbc24be92548bdafa8f13dab1a00e6a42b3e3.

As for your second question, currently the `Reader` doesn't expose any filtering options.

karlding, 2020-05-22:
Closing since the fix is released as part of [v3.3.3](https://github.com/hardbyte/python-can/tree/3.3.3).

closed 2020-05-22 by karlding