hardbyte/python-can #823: Example: crc.py, AttributeError: 'list' object has no attribute 'channel'

Sicksym, 2020-04-30:
There is a problem with compiling crc.py. 

```
  for counter in range(16):
        checksum = compute_xbr_checksum(msg, counter)
        msg.data[7] = counter + (checksum << 4)
        messages.append(
            can.Message(arbitration_id=msg.arbitration_id, data=msg.data[:])
        )
```

In this for loop the messages are appending itself -> Error.

Is there any chance by changing the message for each transmit in order to append the CRC correctly?

karlding, 2020-05-02:
You're seeing this exception right?

```
Traceback (most recent call last):
  File "examples/crc.py", line 82, in <module>
    crc_send(BUS)
  File "examples/crc.py", line 32, in crc_send
    task = bus.send_periodic(messages, 1)
  File "/home/karl/Documents/Code/python-can/can/bus.py", line 205, in send_periodic
    task = self._send_periodic_internal(msg, period, duration)
  File "/home/karl/Documents/Code/python-can/can/interfaces/socketcan/socketcan.py", line 610, in _send_periodic_internal
    bcm_socket = self._get_bcm_socket(msg.channel or self.channel)
AttributeError: 'list' object has no attribute 'channel'
```

This is because #610 is missing from the `3.3.2` release (which I presume is what you're using), which adds support for arrays of `Message` in interfaces that support this natively (SocketCAN via BCM). The example is written for what's currently in `develop`, using an array of `Message`s. We didn't have this example in [`3.3.2`](https://github.com/hardbyte/python-can/tree/3.3.2.final/examples), which seems to be the confusion. Prior to this, a `Task` mapped 1:1 to a periodic message. Essentially you're trying to use newer features on an older version.

We can potentially request for this feature to be ported over to the `3.3.3` release in #715, as the original plan for these releases was maintenance only (ie. no new features) until the new library major version was released. Not sure if that's still the plan, @hardbyte?

Sicksym, 2020-05-04:
Hey @karlding,

yes this is the exception, thank you for your answer. I am using 3.3.2.

Is there any chance to modify the periodic message from `send_periodic()` every time before it gets send in `3.3.2`? Is there any TX request flag which i could check?

Thank you very much in advance!

karlding, 2020-05-09:
Modifying the periodic `Message` being sent every tick is not supported at the moment. This is because currently, the semantics that `send_periodic` aims to provide are what the [SocketCAN BCM provides](https://www.kernel.org/doc/Documentation/networking/can.txt). In addition, this isn't really a feature supported by most interfaces (or their drivers), so it would have to be done in userspace.

You can implement this yourself using a separate thread, essentially doing something similar to what `ThreadBasedCyclicSendTask` does.

However, this has come up a few times in the past, and there's a PR currently being worked on ([829](https://github.com/hardbyte/python-can/pull/829)) that is trying to address parts of this by turning the `data` field a generator (although I make no guarantees about whether the change will ultimately be accepted).

karlding, 2020-09-19:
Closing this out as there doesn't seem to be anything actionable here and the initial query is resolved. Please reopen if necessary.

closed 2020-09-19 by karlding