hardbyte/python-can #836: View virtual CAN messages on CANoe

gcShyam, 2020-05-12:
I am trying to implement a rest bus simulation using the library.
To debug if the code is correct, I use the trace window in CANoe application to check the messages.
Currently, I am trying to send CAN messages on the virtual CAN bus.

The following is the starting code which is supposed to send a message on the virtual CAN.
`
import can
def send_one():
    with can.interface.Bus() as bus:
        bus = can.interface.Bus(bustype='virtual', app_name='SelfApp', channel=[0])
        msg = can.Message(arbitration_id=0xC0FFEE, data=[0, 25, 0, 1, 3, 1, 4, 1], is_extended_id=True)

        try:
            bus.send(msg)
            print(f"Message sent on {bus.channel_info}")
        except can.CanError:
            print("Message NOT sent")
if __name__ == "__main__":
    send_one()
`

I get the following error:
`Traceback (most recent call last):
  File ".../SendMsg.py", line 51, in <module>
    send_one()
  File ".../SendMsg.py", line 30, in send_one
    with can.interface.Bus() as bus:
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\python_can-3.3.2-py3.6.egg\can\interface.py", line 108, in __new__
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\python_can-3.3.2-py3.6.egg\can\util.py", line 197, in load_config
NotImplementedError: Invalid CAN Bus Type - None`

Can someone please help me to remove this error?

bessman, 2020-05-13:
You should create the bus when you open the `with` block, not inside it:
```
import can
def send_one():
    with can.interface.Bus(bustype='virtual', app_name='SelfApp', channel=0) as bus:
        msg = can.Message(arbitration_id=0xC0FFEE, data=[0, 25, 0, 1, 3, 1, 4, 1], is_extended_id=True)
        try:
            bus.send(msg)
            print(f"Message sent on {bus.channel_info}")
        except can.CanError:
            print("Message NOT sent")
if __name__ == "__main__":
    send_one()
```
Also, note that `channel` cannot be a list.

gcShyam, 2020-05-13:
Now I get the following message.
`Message sent on Virtual bus channel 0`
But still, I am unable to see the message on the trace window of CANoe application.

bessman, 2020-05-13:
A `virtual` bus exists only inside python-can and cannot communicate with external interfaces or applications. You probably want `bustype="vector"`, in which case `channel` **should** be a list.

gcShyam, 2020-05-13:
I did try the `bustype="vector"` initially.
But I was receiving the following error.
`can.interfaces.vector.exceptions.VectorError: xlGetApplConfig failed (XL_ERROR)`

P.S. - I have installed the XL Driver Library provided in the Vector's website.

bessman, 2020-05-13:
I don't personally use Vector hardware so I'm not familiar with the specifics of that part of python-can. But the [documentation](https://python-can.readthedocs.io/en/master/interfaces/vector.html) mentions that you must use Vector Hardware Config to specify an application name. Have you done so for your "SelfApp"?

gcShyam, 2020-05-14:
Yes. It worked after I added the application name in the Vector Hardware Config. Thank you.

Another question: There seems to be an API in the XL Driver Library - `xlSetApplConfig`.
Is there support for using this API in python-can?

zariiii9003, 2020-05-14:
Not yet, i will create a pull request for that soon.

closed 2020-05-22 by hardbyte

torn4, 2022-03-04:
Did you have to change any settings inside of CANoe to be able to see can messages sent from python inside of "CANoes Trace window" ?