hardbyte/python-can #838: How can record the TX(send) message in can.logger()

hmq2018, 2020-05-18:
Using the following code, the program can only log the received message to the log.
How to record the TX message into the log, thank you!

```
        self.bus = can.interface.Bus(bustype=self.interface, channel=self.channel, bitrate=self.bitrate, PCAN_BUSOFF_AUTORESET=True) 
        self.logger = can.Logger(test.asc)
        self.notifier = can.Notifier(self.bus, [self.logger, can.Printer()])
```

tamenol, 2020-05-28:
`self.logger.info(msg)` in your send function? The thing is that you have control when sending a message, so appending that message to a log somewhere should be trivial.

hmq2018, 2021-02-15:
Thanks.
But how can use can.Printer() to print RX and TX msg in same time?

christiansandberg, 2021-02-15:
Some interfaces support the `receive_own_messages` argument but PCAN unfortunately does not seem to have a way to accomplish this.

It needs to be done by the driver so that order and timestamp is correct.

hmq2018, 2021-02-16:
@christiansandberg 
Thanks for your explanation

closed 2021-02-16 by hmq2018