hardbyte/python-can #843: can.viewer is not displaying CAN-FD messages

hdnauth, 2020-05-26:
In my test environment,
I have 2 can channels with using Peak-CAN hardware on RaspberryPi. (socketCAN)
can0 is CAN-FD (500K -> 2000Kbps), and can1 is HS-CAN (500Kbps)

first, I tested using cantools to confirm both can0&can1 are woking.

belows are test results on my environment.

input:
`candump can1`

output:
`  can1  660   [8]  60 48 09 00 54 14 00 00
  can1  661   [8]  02 00 00 00 00 00 00 00
  can1  562   [8]  62 04 00 00 00 00 FF FF
  can1  660   [8]  60 48 09 00 54 14 00 00
  can1  661   [8]  02 00 00 00 00 00 00 00
  can1  660   [8]  60 48 09 00 54 14 00 00
  can1  661   [8]  02 00 00 00 00 00 00 00
  can1  660   [8]  60 48 09 00 54 14 00 00
  can1  661   [8]  02 00 00 00 00 00 00 00
  can1  660   [8]  60 48 09 00 54 14 00 00
  can1  661   [8]  02 00 00 00 00 00 00 00
  can1  662   [8]  41 30 31 31 41 31 38 00`

input:
`candump can0`

output:
`  can0  29A  [08]  31 2B 13 00 00 40 00 00
  can0  29A  [08]  E2 6C 14 00 00 40 00 00
  can0  29A  [08]  FA AB 15 00 00 40 00 00
  can0  29A  [08]  29 EC 16 00 00 40 00 00
  can0  29A  [08]  5C 24 17 00 00 40 00 00
  can0  29A  [08]  8F 63 18 00 00 40 00 00
  can0  29A  [08]  2B 2A 19 00 00 40 00 00
  can0  29A  [08]  9E FD 1A 00 00 40 00 00
  can0  29A  [08]  EB 35 1B 00 00 40 00 00
  can0  29A  [08]  38 72 1C 00 00 40 00 00
  can0  29A  [08]  20 B5 1D 00 00 40 00 00
  can0  29A  [08]  F3 F2 1E 00 00 40 00 00`

input:
'python3 -m can.viewer -i socketcan -c can1'

output:
`Count   Time           dt          ID          DLC  Data
10      1.800458       0.200031    0x660       8    60 48 09 00 54 14 00 00
10      1.805534       0.199150    0x661       8    02 00 00 00 00 00 00 00
2       1.010172       1.000313    0x662       8    41 30 31 31 41 31 38 00
2       1.336196       1.001216    0x562       8    62 04 00 00 00 00 FF FF`

input:
`python3 -m can.viewer -i socketcan -c can0`

output:
`Count   Time           dt          ID          DLC  Data`

no output messages are displayed on CAN-FD channel.

christiansandberg, 2020-05-26:
You also need to add `--fd` and `--data_bitrate` switches to activate and configure CAN-FD.

```sh
python3 -m can.viewer --help
```

hdnauth, 2020-05-26:
@christiansandberg 
both --fd and --data-bitrate 2000000 switches don't work for me.
also, I have tried -b 2000000 but no difference too.
(anyway, I don't understand why -b option exist, even though the bus speed is already set on socket configuration)

christiansandberg, 2020-05-27:
You're right of course, you don't need to specify bitrates for SocketCAN. Do you get an error message or you just don't see any traffic when you add `--fd`?

hdnauth, 2020-05-28:
when enter command,
`python3 -m can.viewer -i socketcan -c can0 --fd`

below messages are shown,
`Usage: python -m can.viewer [-h] [--version] [-b BITRATE] [-c CHANNEL]
                            [-d {<id>:<format>,<id>:<format>:<scaling1>:...:<scalingN>,file.txt}]
                            [-f {<can_id>:<can_mask>,<can_id>~<can_mask>}]
                            [-i {canalystii,iscan,ixxat,kvaser,neovi,nican,pcan,serial,slcan,socketcan,socketcan_ctypes,socketcan_native,systec,usb2can,vector,virtual}]
python -m can.viewer: error: unrecognized arguments: --fd`

christiansandberg, 2020-05-28:
Install from develop instead:

```sh
pip3 install https://github.com/hardbyte/python-can/archive/develop.zip
```

hdnauth, 2020-05-29:
thanks @christiansandberg,
It works well now.
anyway, why CAN-FD is supported only on develop version?

christiansandberg, 2020-05-31:
The 4.0 release has been delayed mostly due to dropped support for Python 3.5 but that was mostly an issue back then. Now Debian stable has a newer version so I'm hoping for at least a beta release soon.

closed 2020-06-03 by christiansandberg