hardbyte/python-can #857: BLFParseError

AllenGHL, 2020-06-12:
hi,
    I need to read a canoe created blf file , but the feedback is :
  File "D:\MyTool\blf2asc\test\blf2asc.py", line 35, in <module>
    for msg in can_log:
  File "D:\ProgramFiles\anaconda3\lib\site-packages\can\io\blf.py", line 195, in __iter__
    raise BLFParseError()
can.io.blf.BLFParseError

the blf file could convert asc by canoe tool sucessfully.
I am blocked by this problem for a long time
thanks 
[attach file is the blf file]

AllenGHL, 2020-06-12:
this is my first comment in the github, I'm new , I don't know how could I sent the error blf file to you? How could I upload the file ?  Could you give me a e-mail? 
Thanks

AllenGHL, 2020-06-12:
I upload the zip file ,but I cannot see it

AllenGHL, 2020-06-12:
[https://github.com/AllenGHL/blf-file
[test.zip](https://github.com/hardbyte/python-can/files/4770099/test.zip)
](url)

AllenGHL, 2020-06-12:
if the attach file has problem , I also upload my repository 
https://github.com/AllenGHL/blf-file

christiansandberg, 2020-06-12:
Have you tried with develop?

```sh
pip install https://github.com/hardbyte/python-can/archive/develop.zip
```

AllenGHL, 2020-06-13:
> 

I tried the develop version ,  it worked. Thank you for your reply so quickly. Thank you , thank you.

closed 2020-06-13 by christiansandberg