hardbyte/python-can #859: Filter documentation improvements

hardbyte, 2020-06-18:
Needs a bit more explanation and examples.

Links:
- http://www.cse.dmu.ac.uk/~eg/tele/CanbusIDandMask.html

felixdivo, 2021-04-23:
Doesn't [this](https://python-can.readthedocs.io/en/develop/bus.html#can.BusABC.set_filters) suffice?

hardbyte, 2021-04-25:
It is certainly better than it was but still no, not really. It serves as a good reference for those familiar with the library. If I recall someone asked for some examples of using can filters - including what happens to example messages.

More generally python-can could really do with a few [documentation guides](https://docs-guide.readthedocs.io/en/latest/doctypes/) to compliment the example code and reference docs.