hardbyte/python-can #861: Socket-can

stefan-mrtc, 2020-06-22:
With socket-can, if messages are received too quickly, the receive order is scrambled.
Python 3.7.3 (default, Dec 20 2019, 18:57:59)
Hardware : Raspberry pi 3b+ with CAN hat

------------CODE------------
CAN_SOCKET = can.interface.Bus(channel = "can0", bustype="socketcan")
input_sockets = [CAN_SOCKET]
output_sockets = [CAN_SOCKET]
while (1):
	readable, writable, exceptional = select.select(input_sockets, output_sockets, input_sockets)
	for s in readable:
		if s == CAN_SOCKET:
			message = CAN_SOCKET.recv()

Sending from Microcontroller:
CAN Received (108, 253, 187, 170, 19, 8, 14, 251)
CAN Received (109, 234, 190, 135, 251, 126, 126, 212)
CAN Received (110, 226, 95, 97, 226, 227, 114, 187)

Receiving: 
Result using python socket:
CAN Received (108, 253, 187, 170, 19, 8, 14, 251) from 1094
CAN Received (110, 226, 95, 97, 226, 227, 114, 187) from 1094
CAN Received (109, 234, 190, 135, 251, 126, 126, 212) from 1094

Received with National Instruments CAN Hardware
(This proves the sending is not out of sync)
CAN Received (108, 253, 187, 170, 19, 8, 14, 251)
CAN Received (109, 234, 190, 135, 251, 126, 126, 212)
CAN Received (110, 226, 95, 97, 226, 227, 114, 187)

karlding, 2020-07-17:
Hmm, the Pi Hat uses the MCP2515 CAN Controller right? I don't think `python-can` does anything special beyond reading the socket for SocketCAN..

Out of curiosity, do you also see this when using `candump`?

hardbyte, 2020-07-24:
Can you see if it occurs using just one `select` (there is one in the `bus.recv` call for socketcan), I'd just do something like this:

```python
      try:
            while True:
                msg = bus.recv(1)
                if msg is not None:
                    print(msg)

        except KeyboardInterrupt:
            pass  # exit normally
```