hardbyte/python-can #862: Vector CANoe Virtual Channel

Matthis-wilh, 2020-07-01:
Hi, 
I´ve got a question / issue? 
I use python-can with Vector CANoe to log the CAN Data and use it in python. When I´m using a real CAN-Interface (VN1610), I can view the traffic with the can.viewer. 
How ever I´m not able to configure the pyhton-can can.viewer to show me messages send on the virtual can channel of CANoe with the interface not connected.
When I start the can.viewer the CANalyzer is shown as active in the hardware config, but CANoe isn´t shown as active during the measurement, even if the CANoe trace window shows that messages are sent.

Is this caused by python-can or by CANoe not supporting "virtual can"?

Here´s my hardware config from CANoe
[image]

christiansandberg, 2020-07-01:
How do you start can.viewer?

Matthis-wilh, 2020-07-01:
`python -m can.viewer -i vector -c 0,1`

pierreluctg, 2020-07-02:
Vector Virtual CAN is supported. Can you send frames from python-can (i.e. using can.player) and see the frames in can.viewer running in another process?

Matthis-wilh, 2020-07-03:
Yes.
I started can.viewer in a seperate window with the options mentioned in my last comment and then run the following python script:
`import can
bus = can.interface.Bus(bustype='vector', channel=0, bitrate=250000)
msg = can.Message(arbitration_id=0xa2,
                   data=[1, 2, 3, 4, 5, 6, 7, 8],
                   extended_id=False)
bus.send(msg)`

The viewer shows the message. 
I´m believe the issue is caused by CANoe. Unfortunately I´m pretty new to CANoe so I´ve no idea whats the problem.

FredericChang, 2021-05-21:
As I know when you saw one of channel is acvtive, it means you're running CANOE/CANAPE or CANalyzer and those software occpied virtual channel already.
Note:
Vector product allows multi application to control their device at the same time.

As you know PythonCAN support virtual channel, you could simulate it to test. 
PythonCAN-->Virtual Channel 1 --> use logger mode to listen message.
CANOE/CANAPE or CANalyzer --> Virtual Channel 1 --> Send message.