hardbyte/python-can #864: Set filter on two can IDs

Ahmed94Dawy, 2020-07-02:
Hello,

I am trying to do filter some can messages from a vehicle bus, how can i manipulate the function set filters to accept two can ids instead of one?.

for example:

        can_filters = [{"can_id": 0x101, "can_mask": 0x1FF, "extended": False}]
        bus.set_filters(can_filters)

How do I filter on 0x101 and 0x11e and ignore the rest?

Thanks in advance

RehabHsn, 2020-07-02:
can_filters = [{"can_id": 0x101, "can_mask": 0x1FF, "extended": False}, {"can_id": 0x11e, "can_mask": 0x1FF, "extended": False}]
 bus.set_filters(can_filters)

karlding, 2020-07-16:
> the problem is sometimes there is some message comes with DLC 6 which isn't related to the data sent on the bus, the message original DLC is 8 and it is printed correctly

It's not clear to me what you're saying here.

Are you saying that only 8-byte messages are being sent and there's a bug in the library such that it is returning 6-byte messages?

Or are you looking for a way to distinguish between them? If that's the case, that's dependent on the protocol,and you might want to look at decoding CAN messages via a DBC using [cantools](https://github.com/eerimoq/cantools/) or something.

CSG2019, 2020-08-27:
I also have the similar question, how to extract the specific message ID and data. who can help to provide a detailed example?
in this case, can_filters = [{"can_id": 0x101, "can_mask": 0x1FF, "extended": False}] does it mean i can only get the 0x101 message from CAN bus? what's the difference with can_filter = [{"can_id": 0x101}]?
thank you.

karlding, 2020-09-19:
You might want to look at the `python-can` [documentation](https://python-can.readthedocs.io/en/3.3.3/bus.html#can.BusABC.set_filters) for `set_filters`, which explains the semantics that the filter provides. This behaviour matches what most hardware interfaces implement. You may find the [Linux SocketCAN API](https://www.kernel.org/doc/Documentation/networking/can.txt) documentation about filtering useful if that's easier for you to understand, and [this page](http://www.cse.dmu.ac.uk/~eg/tele/CanbusIDandMask.html) provides some examples.

In your example, you have `can_id = 0x101`, `can_mask = 0x1FF`, and `extended = False`. Thus, this accepts CAN frames whenever the following is true:

```
((received can id) & (can mask)) == ((can id) & (can mask))
((received can id) & 0x1FF) == (0x101 & 0x1FF)
```

In other words, it matches whenever it's a standard frame where the lower 9 bits of the Arbitration ID are equal to `0b100000001`.

For your second question, `python-can` requires you to specify a CAN ID and a CAN Mask, while the `extended` key is optional. So the code would raise a `KeyError` since the caller has not provided a `dict` with the correct key set.

karlding, 2020-09-19:
Closing this out as there doesn't seem to be anything actionable here. Please reopen if necessary.

closed 2020-09-19 by karlding