hardbyte/python-can #866: Report "XL_ERR_WRONG_PARAMETER" error

mmclwd, 2020-07-10:
In my code, I will send APP messages and UDS messages(with udsoncan and iso-tp library) with vector hardware.
While running for about 6h, my tool will report this error, and at this time UDS messages missing but APP messages still transmited.
I don't know the exact reason regarding this "XL_ERR_WRONG_PARAMETER" error, and if capture this error how to resolve this issue to make my code continue working normally?
I already checked vector handbook, it seems that there is only one function - "xlMost150StreamGetInfo" will return this error, but I didn't find this function in python-can library.

zariiii9003, 2020-07-12:
The VectorError tells you which function causes the error. Can you post the traceback here?
Which python-can version are you using?

mmclwd, 2020-07-23:
I add reset CAN in canlib.py and do some other modification, this issue missing.

zariiii9003, 2020-08-27:
@mmclwd did you solve your problem?

DanielTian90, 2020-09-08:
Yes, this issue was fixed, many thanks.

closed 2020-09-17 by zariiii9003