hardbyte/python-can #867: SocketCAN Kvaser no response

kbiesiadecki141, 2020-07-10:
Hello,

I am trying to communicate with a device over CAN using python-can. I have been able to successfully establish communication in both directions on Windows (using Kvaser's canlib) and am trying to do the same on Linux (Ubuntu 18.04) with python-can. 

The messages I'm sending have the correct arbitration IDs and I'm not using CAN FD.

` bus = can.interface.Bus(bustype='socketcan', channel='can0', bitrate=500000, receive_own_messages=True)
`

Some additional information is that when using 'kvaser', it says that my device is not found. I suspect that could've been part of the issue, and I also do not know how to fix this. However, when I run the same program on Windows and the device IS found, there is still no bidirectional communication. (The success with Windows was a result of using Kvaser's canlib, apparently.)

Using can.logger, I can see my messages supposedly getting sent out properly, but I am not receiving any of the expected messages back. Any thoughts/ways I could troubleshoot this? Thanks in advance!

karlding, 2020-07-17:
Since it looks like you're using a SocketCAN driver, you can try running `candump`/`cansend` on the interface to see if the messages are being received/transmitted. That'll help narrow down where the issue is.

kbiesiadecki141, 2020-07-17:
Thanks for the response! Using candump, I can see that my messages are getting sent out on can0, but no messages are being received back as expected. (I know what messages I should be expecting, too, since I verified these on Windows.)

So, for example, this is a command I'm sending out:
  can0  37F   [4]  50 58 00 00

but with no response.

karlding, 2020-07-19:
How are you initializing `can0`? I assume you're doing something like this?

```bash
sudo ip link set can0 type can bitrate 5000000
sudo ip link set up can0
```

If you're seeing the same problem with `candump`, this doesn't seem to be an issue with `python-can` since it just reads from the `can_raw` socket. I would check to ensure that the kernel module is being loaded and you've configured your network correctly.

It's also not exactly clear to me whether you're using CANLib drivers or SocketCAN drivers. Kvaser's Linux drivers blacklist SocketCAN drivers, so if you've installed that in the past, you will need to uninstall so the driver is no longer blacklisted.

kbiesiadecki141, 2020-07-21:
Yes, I've loaded the kernel modules and initialized can0 as you did. I did not install Kvaser's Linux drivers.

DustinMoriarty, 2020-10-22:
Just to double check. Have you made sure that you do NOT have any of the kvaser software installed? The kvaser linuxcan software will specifically break socketcan. Also, you mentioned you tried windows and are now trying linux. By chance are these the same machine with one a VM on the other?

DustinMoriarty, 2020-10-22:
Are you sure that the kvaser device you are using supports 500 kbd? Not all of them do. The newer leaf modules do.