hardbyte/python-can #868: can.interfaces.pcan.pcan.PcanError: A PCAN Channel has not been initialized yet or the initialization process has failed

catmemo, 2020-07-13:
Hi, I tried to use python-can to wakeup a device by PCAN but failed.
But I found nothing wrong when I compared with the examples:
**self.bus = can.interface.Bus(bustype='pcan', channel = 'PCAN_USBBUS'+ str(int(self.channel)+1), bitrate = int(self.baudRate))**
Could anyone help to check? Thanks in advance!

Traceback (most recent call last):
  File "D:\General\Learning\Ifan\Programme\python-can\CAN_Tool.py", line 251, in onStart
    self.bus = can.interface.Bus(bustype='pcan', channel = 'PCAN_USBBUS'+ str(int(self.channel)+1), bitrate = int(self.baudRate))
  File "D:\AIDA\Test_Tools\Python\lib\site-packages\can\interface.py", line 127, in __new__
    return cls(channel, *args, **kwargs)
  File "D:\AIDA\Test_Tools\Python\lib\site-packages\can\interfaces\pcan\pcan.py", line 198, in __init__
    raise PcanError(self._get_formatted_error(result))
can.interfaces.pcan.pcan.PcanError: A PCAN Channel has not been initialized yet or the initialization process has failed

bmeisels, 2020-07-23:
Hi @catmemo ,
have you installed the PCAN drivers and the PCANBasic API DLLs?

catmemo, 2020-07-25:
> Hi @catmemo ,
> have you installed the PCAN drivers and the PCANBasic API DLLs?

Hi Bmeisels,

Thanks for your reply.
I have installed the PCAN drivers.
But I did not install the  PCANBasic API DLLs. 
I'll try and feedback to you, thks again!

bmeisels, 2020-07-27:
Yes.
Have you opened the same PCAN channel with another tool?

tamenol, 2020-08-14:
AFAIK this is no bug in python-pcan. I sometimes get this error too, and it's due to I have the channel open in PCAN-View or my PCAN object didn't get closed/deleted properly.

catmemo, 2020-09-17:
> Yes.
> Have you opened the same PCAN channel with another tool?

Yes, this PCAN tool was used with PCAN-Exploer5 but not at the same time.

catmemo, 2020-09-17:
> AFAIK this is no bug in python-pcan. I sometimes get this error too, and it's due to I have the channel open in PCAN-View or my PCAN object didn't get closed/deleted properly.

Thank you very much! I open the PCAN channel with PCAN-Exploer and closed it again, then it's normal with python-can.
Is that the root cause?

closed 2020-10-21 by catmemo