hardbyte/python-can #871: Align bus.recv(timeout=n) method to raise an exception rather than return None when the timeout period is exceeded.

SoloITBiz, 2020-07-16:
Returning None when the timeout period has been exceeded seems to be inconsistent with other python methods that accept timeout parameters:-

I propose that a TimeoutError is raised instead of returning None:
*** Python Docs - https://docs.python.org/3/library/exceptions.html#TimeoutError
     exception TimeoutError
     Raised when a system function timed out at the system level. Corresponds to errno ETIMEDOUT
***
This would not be backwards compatible, but could be worked around by adding an instance attribute 'raise_timeout_exceptions' that defaults to False and so returns None or may be set to True in which case an exception is raised.

hardbyte, 2020-07-26:
This seems reasonable, and could be part of the 4.0 release. 

Do you want to open a PR?

christiansandberg, 2020-07-26:
I think it is more for keeping the application alive even when no messages arrive at the moment which is perfectly normal and usually not considered a fault.

You would have to change every interface if it should be implemented correctly. Also it would break a lot of implementations out there relying on today's behavior but only at certain edge cases which might lead to frustration when applications suddenly crash unexpectedly. Not everyone will read the changelog carefully and packages which depend on python-can may not be very restrictive on major version numbers.

Maybe the name "timeout" is misleading and should be renamed "max_wait" or something? We should allow a few versions of deprecation warnings before changing such a major feature.

hardbyte, 2020-08-05:
Yeah not receiving a message isn't _exceptional_, I hadn't considered how disruptive a change it would be for users.

closed 2020-08-05 by hardbyte