hardbyte/python-can #872: Notifier timeout parameter

HIllerrr, 2020-07-20:
I'm trying to understand whats the objective of the ` timeout` parameter on the initialization of `can.Notifier` object. 
I would like to raise a flag or exception if I do not receive any messages in a specific time window. But seems that nothing happens if I don't get any message.

Tank you.

zariiii9003, 2020-07-21:
The Notifier passes the `timeout` to the `recv()` method of the bus so it does not block forever. If a timeout occurs the `recv()` method returns and is called again by the Notifier. No Exceptions are raised.

If you need a global timeout (and if you do not use asyncio with a file descripter) you could subclass `can.Notifier` and reimplement the ` can.Notifier._rx_thread()` method.

HIllerrr, 2020-07-22:
Ok, got it.
Thank you so much for your attention and your time.

closed 2020-07-22 by HIllerrr