hardbyte/python-can #874: How to record CAN log with several shorter pieces

mmclwd, 2020-07-23:
If run python-can for a long time to record CAN trace with ASCWriter, size of asc file is extremely large(more than 5GB), and hard to open and analyze by other tools.
Is there any easier way to record long term asc file with several files with specific size like 1GB, and not missing any CAN messages.

hardbyte, 2020-07-23:
It should be relatively straightforward to write something like a `RotatingFileWriter` - it could be a good addition to the `can.io` module if you wanted to open a PR. 

Suggest it implements the [`Listener`](https://github.com/hardbyte/python-can/blob/develop/can/listener.py#L22) interface and just passes the messages on to an [`ASCWriter`](https://github.com/hardbyte/python-can/blob/develop/can/io/asc.py#L218) instance closing and starting a new ASCWriter when your wrap condition is met.

Let us know how you get on

zariiii9003, 2020-10-03:
Closing this because of merge to the develop branch. Feel free to reopen if there are any issues.

closed 2020-10-03 by zariiii9003