hardbyte/python-can #878: Can't select interface in scripts

mamert, 2020-07-29:
When I create and receive on a bus, it works fine:
```bus = can.interface.Bus(bustype='vector', app_name='CANoe', channel=0, bitrate=500000)```
or
```bus = can.interface.Bus(bustype='vector', serial=HW_SN, channel=2, bitrate=500000) # HW_SN defined elsewhere```

However,  the [scripts](https://python-can.readthedocs.io/en/master/scripts.html) don't work for me:
```python -m can.viewer -i vector -c 1 -b 500000```
gives:
<pre>Traceback (most recent call last):
  File "C:\Python36-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Python36-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python36-32\lib\site-packages\can\logger.py", line 120, in <module>
    main()
  File "C:\Python36-32\lib\site-packages\can\logger.py", line 97, in main
    bus = Bus(results.channel, **config)
  File "C:\Python36-32\lib\site-packages\can\interface.py", line 127, in __new__
    return cls(channel, *args, **kwargs)
  File "C:\Python36-32\lib\site-packages\can\interfaces\vector\canlib.py", line 136, in __init__
    "xlGetChannelIndex")
can.interfaces.vector.exceptions.VectorError: xlGetChannelIndex failed (XL_ERR_HW_NOT_PRESENT)</pre>
Same for -c 0. Other values give:
<pre>...
File "C:\Python36-32\lib\site-packages\can\interface.py", line 127, in __new__
    return cls(channel, *args, **kwargs)
  File "C:\Python36-32\lib\site-packages\can\interfaces\vector\canlib.py", line 125, in __init__
    hw_channel, vxlapi.XL_BUS_TYPE_CAN)
  File "C:\Python36-32\lib\site-packages\can\interfaces\vector\vxlapi.py", line 206, in check_status
    raise VectorError(result, xlGetErrorString(result).decode(), function.__name__)
can.interfaces.vector.exceptions.VectorError: xlGetApplConfig failed (XL_ERROR)</pre>

I am confused that the scripts take neither app name, nor serial.

mamert, 2020-07-29:
It seems the scripts take the FIRST available app config for the specified backend.
In my case, it defaulted to CANalyzer (while I was using CANoe), and CANalyzer's logical interfaces were not configured in Vector Hardware Config.

Setting them helped, but it is a workaround, not a solution.

karlding, 2020-08-05:
Right, this is because `app_name` and `serial` are Vector-specific arguments in the initializer, and the script currently doesn't actually provide a way to account for those options. We might want to consider adding a way to do so? I'm not exactly sure what the original intended scope of these scripts were, but perhaps @Lauszus can maybe provide some insight.

And as you mentioned, it takes the default channel configuration for CANalyzer, which is [documented in the docs](https://python-can.readthedocs.io/en/3.3.3/interfaces/vector.html) for the Vector interface, but perhaps that can be made more clear in the script documentation.

Lauszus, 2020-08-20:
@karlding it should be fairly simply to add those Vector specific argument just like when CAN-FD was added to the script: https://github.com/hardbyte/python-can/commit/b1c7c131e63df182cf63c375c1cff713187cdd8a#diff-dad5644a0e09eac1a7acbf60b62d4eeb

felixdivo, 2022-02-06:
This should be covered by  #1170

felixdivo, 2022-11-16:
I suppose it is.

closed 2022-11-16 by felixdivo