hardbyte/python-can #882: General Question: convert raw data to a can message

ashwinsarvesh, 2020-08-02:
Hello,

I want to convert raw data into CAN messages of the type [can_msgs/Frame](http://docs.ros.org/melodic/api/can_msgs/html/msg/Frame.html)
These are the data of a moving car in a driving simulator(for eg. speed of the car), so they keep changing instantaneously.
Can I do this conversion in real-time?

hardbyte, 2020-08-04:
That depends what "real time" means to you! As with any computation it will take some time and it will depend on your hardware. From my experience you can easily create can messages that come in every few milliseconds using a laptop or even a raspberry pi.

zariiii9003, 2020-08-04:
If you have a dbc-file for your bus then you could use this python package: https://github.com/eerimoq/cantools.
It supports encoding a dictionary of signals and values into a bytearray that you can send with python-can

karlding, 2020-09-19:
Closing this out as there doesn't seem to be anything actionable here. Please reopen if necessary.

closed 2020-09-19 by karlding