hardbyte/python-can #883: Feature Request: Add J2534 support

jwdsoft, 2020-08-03:
Thanks for this great library which is working great.
I'm asking to add J2534 device support since all of the libraries on github for J2534 are not working (tried all of them) since you are supporting a lot of interfaces in this library why not adding J2534 devices to the list it would be great
Thanks.

christiansandberg, 2020-08-03:
Support can only be added by someone with access to the specific hardware, so you will probably have to implement it yourself. Since this is a CAN only library, the implementation would only support the CAN protocol, but not any of the other protocols supported by J2534.

felixdivo, 2020-08-06:
While this is apparently a useful feature, it might also be well out of scope of this project.

jwdsoft, 2020-08-06:
Thanks for your answer
I tried some libraries found on github one of them sends data but don't receive them.created issues on there github library but the projects are not active and the problem wasn't solved.
Also tried some C# library for j2534 but none of them could send nor receive.(at this point you could say my device is broken but i know it works since i use it with a software and it works fine)
Tried my best to see what is the problem but couldn't find what it is
Wanted to write my own but I was stuck
What I requested is not to make a J2534 library (to handle all j2534 calls from python )
What i meant is this library is usefull and it has a lot of hardware support like slcan or kvaser.i use it with both these two
But sometimes I don't have them around and I have a j2534 device and I wish that python-can had j2534 support (just sending a receiving frames like kvaser and not j2534 specific implementation just to be able to send and receive can bus frames)

jwdsoft, 2020-10-02:
Hello @christiansandberg 
Yes that's what I asked for just pure can implementation not j2534 I asked to add support for J2534 hardware in this can library just like kvaser for example it is supported by python-can and it is a J2534 device compatible.
If J2534 support is added (just sending and receiving pure can bus frames) it will be great since there are a lot of J2534 that will work with this library

christiansandberg, 2020-10-02:
You're welcome to file a pull request with this if you want to!

pyzg, 2023-12-30:
It is recommended to add support for CLASSIC CAN by selecting a J2534device through J2534 PassThruSupport.04.04.

Utilize the following ProtocolID value: 0x00000005; CLASSIC CAN.

without the J1962_PINS parameter.