hardbyte/python-can #886: How can I convert csv format to blf format?

wangtun, 2020-08-06:
as the title , is there a convert function for converting a can file to another format file?  thanks

felixdivo, 2020-08-06:
Hey @wangtun 

You can have a look at this example: (python-can/examples/simple_log_converter.py)[https://github.com/hardbyte/python-can/blob/develop/examples/simple_log_converter.py]. That's pretty much you use case. If you already know the formats, you might consider replacing `can.LogReader` and `can.Logger` with specific readers and writers, which are both documented [here](https://python-can.readthedocs.io/en/master/listeners.html). There are some configuration options via the constructors too. 

Hope that helps. If yes, please close that issue. ;)

Felix

wangtun, 2020-08-17:
@felixdivo hi, thank you very much,  it is helpful for me, thanks

closed 2020-08-17 by wangtun