hardbyte/python-can #889: Message object unexpected interpretation

johanbrus, 2020-08-06:
I'm seeing some unexpected behavior in the way messages are displayed.
Sending 29-bit message 0xC0FFEE with data 0x38 0x1A 0x2B 0xC5 0x8D 0x6E 0x22 0xFF
Receiving a message using "msg2 = bus.recv(1.0)"
Then executing following print commands with following results:

Sending with data field|"print(msg2.data)"|"print(msg2)"
-------|-------------|------------------
0x38|bytearray(b'8')|Timestamp:        0.696006    ID: 00c0ffee    X                DLC:  1    38                          '8'    Channel: 0
0x38 0x1A|bytearray(b'8\x1a')|Timestamp:        2.156239    ID: 00c0ffee    X                DLC:  2    38 1a                       Channel: 0
0x38 0x1A 0x2B|bytearray(b'8\x1a+')|Timestamp:        1.978031    ID: 00c0ffee    X                DLC:  3    38 1a 2b                    Channel: 0
0x38 0x1A 0x2B 0xC5|bytearray(b'8\x1a+\xc5')|Timestamp:        1.915171    ID: 00c0ffee    X                DLC:  4    38 1a 2b c5                 Channel: 0
0x38 0x1A 0x2B 0xC5 0x8D|bytearray(b'8\x1a+\xc5\x8d')|Timestamp:        1.846557    ID: 00c0ffee    X                DLC:  5    38 1a 2b c5 8d              Channel: 0
0x38 0x1A 0x2B 0xC5 0x8D 0x6E|bytearray(b'8\x1a+\xc5\x8dn')|Timestamp:        3.756427    ID: 00c0ffee    X                DLC:  6    38 1a 2b c5 8d 6e           Channel: 0
0x38 0x1A 0x2B 0xC5 0x8D 0x6E 0x22|bytearray(b'8\x1a+\xc5\x8dn"')|Timestamp:        3.561521    ID: 00c0ffee    X                DLC:  7    38 1a 2b c5 8d 6e 22        Channel: 0
0x38 0x1A 0x2B 0xC5 0x8D 0x6E 0x22 0xFF|bytearray(b'8\x1a+\xc5\x8dn"\xff')|Timestamp:        2.771891    ID: 00c0ffee    X                DLC:  8    38 1a 2b c5 8d 6e 22 ff     Channel: 0

As you can see the msg2.data object is unexpected and the msg2 object when DLC=1 has a funny '8' hanging in mid-air.
Any workarounds for time being also welcome.

Same behavior observed with 11-bit (standard) identifier.
I have seen this on both kvaser and IXXAT interface so maybe generic issue.

christiansandberg, 2020-08-06:
The string is printed using ASCII characters if possible. 0x38 is the code for the character "8" for example.

johanbrus, 2020-08-06:
@christiansandberg oh, is that what is happening? But then when the ASCII character doesn't exist, e.g. 0x1A = SUB, the character is somewhat displayed as a hex? It still doesn't seem consistent to me.
However, I am able to extract the values by accessing the different positions in the bytearray (e.g. msg2.data[0]) and extract the decimal values after which I can convert them to hexadecimal.

So this is maybe more like a feature than a bug? In that case just close it then. Thanks for the help!

christiansandberg, 2020-08-06:
It's just how Python prints bytearrays, probably for historical reasons since there previously was no Unicode support and all strings were just bytes.

closed 2020-08-06 by christiansandberg