hardbyte/python-can #890: New Backend Netronics CANdo

johanbrus, 2020-08-06:
Is anyone already working on adding the CANdo CAN interface from Netronics (see www.cananalyser.co.uk/) as a new backend to python-can?

Currently I have a colleague working on this in his free time as we were looking to use this as a CAN interface in my test environment. He's a SW eng but somewhat new to python, hence if someone is willing and able to help with code reviews or to get this backend implemented that would be great. Thus far we are able to configure it and send a single message.

christiansandberg, 2020-08-06:
Make a pull request and we will do our best to review it. There is no one to my knowledge that has access to that hardware so we can't do the implementation.

belliriccardo, 2024-11-05:
Hi to all! I've just submitted #1887, that adds the plugin's interface to the list. The package is already available on [PyPi](https://pypi.org/project/python-can-cando/) and [my GitHub](https://github.com/belliriccardo/python-can-cando). All feedback is welcome!

PR #1887 (2024-11-05): Added Netronic's CANdo and CANdoISO adapters interface
Also closes #890
closed 2024-11-05 by zariiii9003