hardbyte/python-can #893: Monitoring of IXXAT USB-CAN V2 connection

Djentmat, 2020-08-12:
Hi! I'm trying to check connection of my ixxat usb-can. 
I didn't find any function in lib and I had to use things like:

```python
import can

def bus_init()
    try:
        bus = can.interface.Bus(bustype='ixxat', channel=1, bitrate=800000)
    except Exception as err:
        print err

while True:
    time.sleep(1)
    bus_init()
```

This code generates error "**function vciDeviceOpen failed (Ran out of memory)**" after few minutes of work

How can I check my device connection properly? Thx for answer.

johanbrus, 2020-08-19:
I got the same error message with my IXXAT USB-to-CAN compact v1.5 with Python 3.7 and python-can 3.3.3 after a few minutes (about 130 seconds).
Sorry I'm quite new to this but why does this need to run for minutes? Don't you already have the confirmation that it works after a few seconds? What is the use case for such a functionality? Or are you running this in parallel while applying some other things to the bus (with a different device maybe?)?

felixdivo, 2021-04-15:
@Djentmat I don't really get the point of you code, do you want to wait for the bus to open? Anyways, you should do `bus.shutdown()` after you don't need the bus any more. This might have cause the increased resource usage.

What version of Python and the library were you using? Is this still a problem with the version on the develop branch? Please comment again if you need further help and we can re-open the issue.

closed 2021-04-15 by felixdivo