hardbyte/python-can #896: can.logger script doesn't work when started from systemd

paragep, 2020-08-18:
Hi, 

I use a systemd service to log the socketcan can0 with can.logger script.
I use a Unit file that call the script:

...
[Service]
Type=simple
ExecStart=python -m can.logger -f /home/admin/can-logs/test.blf -v -c can0 -i socketcan
...

This script works perfectly when called from the linux shell, but when I start it from the service, the log file is created in /home/admin/can-logs but stay empty.
What could be wrong ? is python-can not able to handle socketcan from systemd service ? 

Thanks for your help,

christiansandberg, 2020-08-18:
Have you set up the SocketCAN interface before the service is started?

paragep, 2020-08-18:
I have a first service that start the can interface can0.
And the can0-logger service is waiting for it to start.
In any case can0 is up and can.logger works from the linux shell, but not when the systemd start the service with exactly the same command.

When I start the logger from the shell I get the message "no handlers could be found for logger can.utils" and then just after the logger start.
When I start the service that calls the same command I get the same error "no handlers could be found for logger can.utils", the file is created but it seems that the logger hangs at this step.

Thanks for your help