hardbyte/python-can #897: IXXAT does not propagate errors

jrast, 2020-08-20:
We have problems with the IXXAT interface: If errors occur on the bus, for example acknowledgment errors, they are logged, but it seems to be impossible to react to them.

How is this handled with other interfaces? Would it be possible to raise an exception if errors occur?

Audry21, 2020-09-09:
I have no idea but have some interest in any hint: I am using the Ixxat interface at work and thought that the python module would be just perfect, bit it lack errors handling: no exception raised if I send a message while nohting is connected to the Ixxat device (which obviously leads to an ack error),. And for recv, error are printed/logged but this is quite useless as there seems to be no way to catch them and react to it.

Any idea about it ? Should I just do my own C/C++ wrapper around the IXXAT API or is there a way to fix the python module ?

flyser, 2020-10-14:
Also see #856