hardbyte/python-can #898: CANoe channels are not working (xlGetApplConfig failed, xlOpenPort failed)

stefanuca, 2020-08-21:
Hello,

I am trying to send a can message from python.
this is the code
[image]

this is vector hardware configuration
[image]
[image]

what might cause the issue?
when i use a virtual bus it works

stefanuca, 2020-08-21:
this is the error

[image]

stefanuca, 2020-08-24:
@zariiii9003 it is possible to have something about the order of the applications in vector hardware configuration?

zariiii9003, 2020-10-03:
@stefanuca Were you able to resolve your problem?

Can you please install this version `pip install https://github.com/zariiii9003/python-can/archive/vector_channel_config.zip` and send me the output from the following code?

```python
import can

[print(cfg) for cfg in can.detect_available_configs(interfaces=['vector'])]
```

felixdivo, 2021-04-23:
@zariiii9003 Can this be closed or do you suspect an error in python-can to be lurking here that needs further digging?

zariiii9003, 2021-04-26:
@felixdivo There might be some bug. #971 also has problems connecting to a piggy interface. I will close this and leave the other one open.

closed 2021-04-26 by zariiii9003