hardbyte/python-can #900: listener = can.Listener()      TypeError: Can't instantiate abstract class Listener with abstract methods on_message_received

wangcunqiang, 2020-08-25:
When I debug the code, I've got a problem in python3.8 below:
    "listener = can.Listener()
     TypeError: Can't instantiate abstract class Listener with abstract methods on_message_received"

I find the explanation as follow:

The ABC module has a metaclass (ABCMeta) and decorators (@AbstractMethod and @AbstractProperty).If an ABC has an @AbstractMethod or @AbstractProperty, it cannot be instantiated but must be overridden within a subclass.

I modify the code in ./python-can-develop/can/listener.py as follow, it's ok:

class ListenerABC(metaclass=ABCMeta):
    """The basic listener that can be called directly to handle some
    CAN message::

        listener = SomeListener()
        msg = my_bus.recv()

        # now either call
        listener(msg)
        # or
        listener.on_message_received(msg)

        # Important to ensure all outputs are flushed
        listener.stop()
    """

    @abstractmethod
    def on_message_received(self, msg: Message):
        """This method is called to handle the given message.

        :param msg: the delivered message

        """

    def __call__(self, msg: Message):
        self.on_message_received(msg)

    def on_error(self, exc: Exception):
        """This method is called to handle any exception in the receive thread.

        :param exc: The exception causing the thread to stop
        """

    def stop(self):
        """
        Stop handling new messages, carry out any final tasks to ensure
        data is persisted and cleanup any open resources.

        Concrete implementations override.
        """

class Listener(ListenerABC):
    """The basic listener that can be called directly to handle some
    CAN message::

        listener = SomeListener()
        msg = my_bus.recv()

        # now either call
        listener(msg)
        # or
        listener.on_message_received(msg)

        # Important to ensure all outputs are flushed
        listener.stop()
    """

    def on_message_received(self, msg: Message):
        """This method is called to handle the given message.

        :param msg: the delivered message

        """

christiansandberg, 2020-08-25:
What do you want to do with your listener? There should not be a need to instantiate a noop listener. You should use e.g. a log file implementation or make your own subclass for processing the messages.

wangcunqiang, 2020-08-26:
When instantiate a listener as follow:

    class CanConnection(object):
        def __init__(self, callback, filter, bus):
            self.__bus = bus
            listener = can.Listener()
            listener.on_message_received = callback
            self.__notifier = can.Notifier(self.__bus, [listener], 0)
            self.__listeners = [listener]
            self.addFilter(filter)

        def addCallback(self, callback):
            listener = can.Listener()
            listener.on_message_received = callback
            self.__notifier.add_listener(listener)
            self.__listeners.append(listener)

    (These codes come from richClubb/python-uds )

The error happened: TypeError: Can't instantiate abstract class Listener with abstract methods on_message_received

Is there a problem with that?

christiansandberg, 2020-08-26:
You can actually use the callback directly as a listener. Pass the callback function directly to the Notifier instead of going through a Listener object.

wangcunqiang, 2020-08-28:
Hi，I get it, thanks.

I have one more question:

    def _on_message_received(self, msg: Message):
        for callback in self.listeners:
            res = callback(msg)
            if self._loop is not None and asyncio.iscoroutine(res):
                # Schedule coroutine
                self._loop.create_task(res)

The "callback" is a listener, then call "callback(msg)" is ok?

felixdivo, 2021-04-22:
Listeners implement the `__call__()` special method (see [here](https://github.com/hardbyte/python-can/blob/404876533b2ec886732869b8a8756018907acae9/can/listener.py#L47-L48)), that's why this works (maybe that was the question?).

Seems like the original issue is resolved then. Please open a new one if there are still problems.

closed 2021-04-22 by felixdivo