hardbyte/python-can #908: Vector canlib: _detect_available_configs method includes non-CAN channels in its return value

dhnc4u, 2020-09-22:
The __detect_available_configs_ method in canlib.py for Vector returns **all** channels enumerated by the driver.
This includes LIN, Digital/Analog I/O, ... channels, which are not in scope of a CAN library, but are supported by the connected hardware.
In addition, I could not find an (implemented) way to get the channel type of the returned channels.

Based on page 63 of the [XL Driver Library Manual](https://assets.vector.com/cms/content/products/XL_Driver_Library/Docs/XL_Driver_Library_Manual_EN.pdf), I propose a minor change to __detect_available_configs_, to only return channels supporting CAN:
`if channel_config.channelBusCapabilities & vxlapi.XL_BUS_ACTIVE_CAP_CAN:`
This requires the following to be added to vxlapi.py:
`XL_BUS_ACTIVE_CAP_CAN = 65536`

Any thoughts or things I missed ?
Thanks, David.

zariiii9003, 2020-09-22:
Hello, it is implemented in the develop branch. See #641

dhnc4u, 2020-09-23:
Thank you for pointing that out @zariiii9003 , I can confirm it is.
This means it will only be available in Python-CAN 4.x, not 3.x, correct ?
What would be the quickest way to have this fix available in a/any stable release please ?

zariiii9003, 2020-09-24:
You can use the 4.0 pre-release: `pip install python-can==4.0.0.dev0`
We could add it to 3.3.4 but that depends on whether @hardbyte considers this a bug or just a missing feature.
I'd say it is the latter.

dhnc4u, 2020-10-03:
Thank you @karlding for the commit.
Slightly off-topic:
Do you know if it ever has been considered to include _transceiverName_ and _serialNumber_ in the dictionary list returned by __detect_available_configs_ ? In our setup we have multiple Vector devices connected (vehicle level tests), and we have a hard time differentiating all the channels it returns. By adding those 2 parameters, the return value would be much clearer.

felixdivo, 2021-04-14:
The original problem has been resolved. There are currently no plans for further 3.x releases (see [here](https://github.com/hardbyte/python-can/releases/tag/3.3.4)). Thanks @dhnc4u and @zariiii9003 for the discussion and fixes.