hardbyte/python-can #909: can not

chuchienshu, 2020-09-24:


chuchienshu, 2020-09-24:
```
    bus = can.interface.Bus(channel=0,
                            bustype="vector",
                            fd=True,
                            error_frame=True) # not sure error_frame is a valid parameter.
    for msg in bus:
        print(msg)
```
I am now wanna check out the CAN FD Error frame. But I can not capture it in python script. any suggestion will be helpful. Thx!

zariiii9003, 2020-10-03:
I don't have an interface available right now, so i didn't test this, but something like this should work with the develop-branch:

```python
from can.interfaces.vector import VectorBus, xlclass, xldefine, xldriver

class MyVectorBus(VectorBus):
    def handle_canfd_event(self, event: xlclass.XLcanRxEvent) -> None:
        print("Handle CAN FD event: ", xldriver.xlCanGetEventString(event))
        
        if event.tag == xldefine.XL_CANFD_RX_EventTags.XL_CAN_EV_TAG_RX_ERROR:
            error_code = event.tagData.canError.errorCode
            error_name = xldefine.XL_CANFD_RX_EV_ERROR_errorCode(error_code).name
            print("XL_CAN_EV_TAG_RX_ERROR: ", error_name)

bus = MyVectorBus(channel=[0], app_name='CANape', bitrate=500000, fd=True)
```

chuchienshu, 2020-10-13:
@zariiii9003 anyway, thanks for you reply~

chuchienshu, 2020-10-20:
@zariiii9003 the script you provided run well with the develop-branch. but I have no idea about how to use handle_canfd_event().
I tried to to something like,
```
bus = MyVectorBus(channel=[0], app_name='CANape', bitrate=500000, fd=True)
while 1:
    rec_msg = bus.recv(1)  # monitor can bus 
    if rec_msg is not None:
        bus.handle_canfd_event(rec_msg)
```
unsurprisely, all I get is: ctypes.ArgumentError: argument 1: <class 'TypeError'>: expected LP_XLcanRxEvent instance instead of Message. I don't know how to construct `event `object for handle_canfd_event().
what I wanna do is to obtain CAN FD ACK Error frame and its timestamp.
any suggestion will be appreciate. thanks a lot.

zariiii9003, 2020-10-23:
You do not call `handle_canfd_event()` directly, it is called by the `recv` method of `VectorBus`. You only have to define what you want to it to do once it is called. Here is how you get the timestamp:

```python
from can.interfaces.vector import VectorBus, xlclass, xldefine, xldriver

class MyVectorBus(VectorBus):
    def handle_canfd_event(self, event: xlclass.XLcanRxEvent) -> None:
        print("Handle CAN FD event: ", xldriver.xlCanGetEventString(event))
        
        if event.tag == xldefine.XL_CANFD_RX_EventTags.XL_CAN_EV_TAG_RX_ERROR:
            timestamp = event.timeStamp * 1e-9 + self._time_offset
            error_code = event.tagData.canError.errorCode
            error_name = xldefine.XL_CANFD_RX_EV_ERROR_errorCode(error_code).name
            print(f"XL_CAN_EV_TAG_RX_ERROR: {error_name}, timestamp: {timestamp}")
```

chuchienshu, 2020-10-24:
@zariiii9003 That do work for me. Many Thanks!!

felixdivo, 2021-04-14:
It is unclear to me what the problem is with the below script. If the problem persists, please open a new issue since the original problem (receiving error frames) is solved.

closed 2021-04-14 by felixdivo