hardbyte/python-can #910: Raise BLFParseError when iterate BLF file with TriggerEvent included

cucdat, 2020-09-25:
Hi,

I get the BLFParseError when trying to iterate the blf file with TriggerEvent included.
If I exclude this event by CANalyzer, I can iterate the file perfectly.
I do not need to read this event but is there any method that we can ignore it automatically and move to the CANbus data.

Below is the TriggerEvent that included in my blf.

[image]

christiansandberg, 2020-09-26:
Which version of python-can are you using? Have you tried with develop?

cucdat, 2020-09-28:
I used the 3.3.3 version and it didn't work.
However, I tried with the develop branch and this issue does not show up anymore.

Thank you so much for your kind support.
I close this open issue.

closed 2020-09-28 by cucdat