hardbyte/python-can #912: CAN line stops with RPI can hat

stefan-mrtc, 2020-09-28:
Hi all.

I am using a raspberry CAN hat using socketcan
https://www.waveshare.com/rs485-can-hat.htm
Baudrate: 10000
RPI kernal:
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: 2f34808bb24246b79ed724449d6b1005
           Boot ID: bc9c1e3ee92a4af29b2739e81d0e2f5e
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 4.19.118-v7+
      Architecture: arm

Everything works ok until I increase the can bandwidth to about 10%
If I increase the can bus bandwidth from about 5 to 10%, 
the RPI stops reading CAN messages.
Upon further investigation, it looks like the RPI stops the clock line going to the MCP25115.
This looks like a deeper problem than the python-can library.
Any assistance would be `appreciated!

Code:

```
#Start CAN Socket
try:
	os.system("sudo ifconfig can0 down")
	os.system("sudo ip link set can0 type can bitrate 1000000")
	os.system("sudo ifconfig can0 txqueuelen 500")
	os.system("sudo ifconfig can0 up")
	#self.CAN_SOCKET = can.interface.Bus(channel = 'can0', bustype = 'socketcan_ctypes')
	self.CAN_SOCKET = can.ThreadSafeBus(channel = "can0", bustype="socketcan")
	print ("CAN connection success!")
except Exception as e:
	print ("CAN connection failed!")
	print (e)
	sys.exit()

#CAN loop back
while (1):
	#if (not self.read_CAN_queue.empty()):
		#can_msg = self.read_CAN_queue.get_nowait()
	can_msg = self.CAN_SOCKET.recv(0.0)
	if (can_msg)
		self.read_CAN_queue.task_done()
		can_tx_message = can.Message(arbitration_id=CAN_TX_ID, data=can_msg.data, is_extended_id=False) 
		self.CAN_SOCKET.send(can_tx_message, timeout=1)

```

qdzpx123, 2020-12-11:
Hi,can you read the CAN frame by the SPI interface.
I am using RPI 3B and Seeed CAN bus is working.

felixdivo, 2021-06-05:
> Upon further investigation, it looks like the RPI stops the clock line going to the MCP25115.
> This looks like a deeper problem than the python-can library.

Sorry, but it seems like we can't help you here. If you still think this is an issue with python-can, please comment again or open a new issue.

closed 2021-06-08 by felixdivo