hardbyte/python-can #919: python-can 3.3.4 breaks functionality in a minor version

albireox, 2020-10-11:
Commit 0c34e5070ccaa3c89d6c1893370083943f0d5390 introduced this change in `message.py`

```python
if key not in self.__slots__:
    raise AttributeError
```

This effectively disables the use of custom attributes in `Message`. It seems that's planned for version 4.0, but this change breaks any code that adds attributes to `Message` and does so in a patch version instead of a major one.

hardbyte, 2020-10-11:
Thanks for reporting, if there is a clean way to still support custom attributes l'm happy to make a quick 3.3.5 release.

I'll note that we documented custom attributes on the Message class haven't been supported since the 3.0 release (See the Message [docs](https://python-can.readthedocs.io/en/3.0.0/message.html#module-can))
> Messages do not support “dynamic” attributes, meaning any others that the documented ones.

@karlding this was introduced from fixing https://github.com/hardbyte/python-can/issues/804, ensuring a message can be pickled. Any ideas for how can we might continue to support both custom attributes and pickling?

albireox, 2020-10-11:
Thanks for the quick reply. For some reason I don't think I've ever seen one of the deprecation warnings, so I wasn't aware of this change. On my side I've updated my code to use a subclass of `Message` in which I override `__slots__` to add my custom attributes, and I expect that will work fine for any version before or after 3.3.4, but I wanted to alert of the problem because I assume other users may be in the same situation.

felixdivo, 2021-04-14:
Is there still some action needed on this?

hardbyte, 2021-04-14:
I don't think any PRs were opened to fix 3.3.4, and there haven't been any other comments in half a year so I think it is safe to close.

closed 2021-04-14 by hardbyte