hardbyte/python-can #922: How to Log CAN FD traffic?

chuchienshu, 2020-10-13:
I'm now wanna Log all traffic on the CAN bus, which including CAN and CAN FD message. However, it seems that [can.logger](https://python-can.readthedocs.io/en/master/scripts.html#can-logger) just record the standard CAN traffic. Is it possible to log CAN FD traffic simultaneously? Any hint will be appraciate. thx~

christiansandberg, 2020-10-15:
You should be able to with develop:

https://python-can.readthedocs.io/en/develop/scripts.html

felixdivo, 2020-11-18:
Also make sure to collect CAN FD messages in the first place. E.g. pass `fd=True` to the Bus constructor.

felixdivo, 2020-11-18:
If you still have problems, feel free to create an issue with the steps you tried and what dd not work as expected.

closed 2020-11-18 by felixdivo