hardbyte/python-can #930: socketcan cannot create multiple periodic messages with same arb id

SiouxerSkate, 2020-11-10:
When using a socketcan device and sending multiple periodic messages I receive the below error

"ValueError: A periodic Task for Arbitration ID XXXX has already been created." Where XXX is the arbitration ID of the message I am trying to send periodically. 

Is it possible to send multiple periodic messages with same ID using socketcan?

I found this issue when running few tests using a systec probe on Linux. On Windows using the systec probe, and the systec option in python-can, I am able to create multiple cyclic messages with the same ID. Running on Linux with the systec probe as a socketcan device the same test fails. I verified with another socket can device as well.

Some context is we will be trying to simulate an ECU on a bus. So its possible that there will be multiple periodic messages coming from the same ID with different data.

ga-dmorey, 2021-01-12:
I ran into a similar issue. It looks to be an issue with socketcan itself. The socketcan [docs](https://www.kernel.org/doc/Documentation/networking/can.txt) mention this under the Broadcast Manager doc (section 4.2) where it says: "The different RX/TX jobs are distinguished by the unique can_id in each BCM message."

tysonite, 2021-03-15:
It shall not be an issue anymore since https://github.com/hardbyte/python-can/pull/785 has been merged.

felixdivo, 2021-04-14:
@SiouxerSkate @ga-dmorey Is this now solved?

felixdivo, 2021-04-16:
I read the referenced PR again. It should indeed be fixed by #785. If not, please comment or open a new issue. Thanks for reporting!

closed 2021-04-16 by felixdivo