hardbyte/python-can #933: Scripts can.logger/can.viewer with peak-system over PCAN_USBBUS1

balumbo, 2020-11-13:
I am trying to use the script can.logger with the PCAN-USB from peak-system. 
But it does not work unfortunately.
When I use the receivetest from Peak-system everything is running.
receivetest -f=/dev/pcanusb32 -b=0x852b -n=500.

When I call the can.logger script with 
python -m can.viewer -b 83333 -c "/dev/pcanusb32" -i pcan
the following message pop up
> Traceback (most recent call last):
>   File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
>     "__main__", fname, loader, pkg_name)
>   File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
>     exec code in run_globals
>   File "/home/airbus/.local/lib/python2.7/site-packages/can/viewer.py", line 502, in <module>
>     main()
>   File "/home/airbus/.local/lib/python2.7/site-packages/can/viewer.py", line 493, in main
>     bus = can.Bus(parsed_args.channel, **config)
>   File "/home/airbus/.local/lib/python2.7/site-packages/can/interface.py", line 127, in __new__
>     return cls(channel, *args, **kwargs)
>   File "/home/airbus/.local/lib/python2.7/site-packages/can/interfaces/pcan/pcan.py", line 173, in __init__
>     self.m_PcanHandle = globals()[channel]
> KeyError: '/dev/pcanusb32'

When I use
python -m can.logger -f /home/balumbo/canlog.txt -c PCAN_USBBUS1 -i pcan -b 83333
The logger started but with no entries in the log file.

> Connected to PcanBus: PCAN_USBBUS1
> Can Logger (Started on 2020-11-12 16:19:47.231058)

The same issue obtain with the can.viewer script.
Which channel has to be used?

> dmesg | grep pcan
> [ 4788.586812] pcan: loading out-of-tree module taints kernel.
> [ 4788.587792] pcan: Release_20200701_n (le)
> [ 4788.587793] pcan: driver config [mod] [isa] [pci] [pec] [dng] [par] [usb] [pcc] 
> [ 4788.593143] pcan 3-1:1.0: PCAN-USB fw v8.4
> [ 4788.651123] pcan: - usb device minor 32 found
> [ 4788.651152] usbcore: registered new interface driver pcan
> [ 4788.651173] pcan: major 239.

felixdivo, 2021-04-15:
Channel 32 seems to just not exist in our library. You can see the available ones [here](https://github.com/hardbyte/python-can/pull/982/files#diff-83cad9b8cc7cd5a75c4020bb5a9b2403d1a960e217e5e2d14c79929224f537f9R76). I don't know PEAK too well, but generally the channels depend on the hardware that you connected.

pkess, 2021-12-19:
@balumbo : It looks like you are using the pcan adapter on linux os. Is this correct? Normally when i use the pcan adapter with linux i am using the socketcan driver provided by peak and integrated in the kernel. With this you can use the `socketcan `backend of `python-can`
Maybe you can try to list the interface with `ip link` and check if there is an interface named `can0` or something similar