hardbyte/python-can #934: timestamp inaccuracy on Windows

danielhrisca, 2020-11-14:
The function ``time.time`` is not reliable for setting the time stamps on Windows. 

Even if the resolution is good enough, the value returned by ``time.time`` is updated according to the interval Windows timer (see https://stackoverflow.com/questions/1938048/high-precision-clock-in-python)

On my PC this defaults to 10ms steps which in my opinion is a very high inaccuracy.

christiansandberg, 2020-11-14:
Which context are you referring to? Message timestamps are derived by hardware when possible, but some interfaces simply don't provide this in which case we use time.time() as a fallback.

danielhrisca, 2020-11-14:
I was looking for example at the Vector implementation were and offset is computed based on ``time.time``

christiansandberg, 2020-11-14:
That's just to convert from relative to absolute timestamps. There will obviously be an absolute error due the fact that we don't read the hardware time and PC time simultaneously but relative accuracy between messages will still be very accurate.

danielhrisca, 2020-11-14:
I my use case I need to synchronize the CAN messages with some other device data, using the PC time as reference. How could I do this as accurately as possible?

christiansandberg, 2020-11-14:
I don't know a better way unfortunately. Maybe you could measure the error somehow and compensate with an offset. Feel free to improve it!

hardbyte, 2020-11-15:
The resolution of `time.time` using a recent Python on a recent windows is <1ms - see https://www.python.org/dev/peps/pep-0564/#windows but if you want to suggest an improvement feel free.

closed 2020-11-15 by hardbyte

danielhrisca, 2020-11-15:
@hardbyte the resolution is good but the value is refreshed at best every ~0.5ms if you change the Windows timer interval. By default it is set at ~10ms. Running the test code from the linked SO thread will show this clearly

hardbyte, 2020-11-15:
Hmm I got ~1ms (not ~10ms) on a windows laptop without changing anything. I'm not convinced we can do much better from python-can?

I ran this:

```python
import time

# measure the smallest time delta by spinning until the time changes
def measure():
    t0 = time.time()
    t1 = t0
    while t1 == t0:
        t1 = time.time()
    return t1-t0

samples = [measure() for i in range(30)]

for s in samples:
    print(f'time delta: {s:.12f} seconds') 
```

I see `time delta: 0.000000238419 seconds` on Linux and `time delta: 0.001001238419 seconds` on Windows.

danielhrisca, 2020-11-15:
@hardbyte @christiansandberg 
I've send PR #936  to improve the accuracy for Vector and Kvaser buses