hardbyte/python-can #935: [3.3.4] Slow library import time

kjngineering, 2020-11-14:
I have been attempting to use python-can on an embedded buildroot system, which essentially will be a CAN based display.
Hardware is relatively low level, single ARM Cortex-A7 1.2GHz, 64mb ram, SPI can interface, SD card based storage/boot media.

I had done some function tests with python-can originally and it had worked as intended, but now I am getting to optimisations and I am stumped. Now I have introduced python-can into my runtime, library import -- "import can" is taking a huge block of time which is interfering with the user experience.

As library imports are blocking (as far as I know) its causing a blank screen for the duration of the load time until the display library can make a call and refresh the display.

My typical load times for the can library are in the 2800ms > 3000ms range, which is forever compared to the rest of the process, some benchmarks for reference:
- Power on to boot screen ~ 1200ms
- Power on to user environment ~1800ms
- Script runtime to user display ~500ms (no import-can)
- Script runtime to user display ~3500ms (with import-can)
- Configure can interface (socket can) start receiving packets once library is loaded (10ms)

I'm using the bare minimum of the library:
- SocketCan interface
- Basic Filters
- Single Listener

No logger, no asnycio,  no other interfaces or  parts of the API.

Is this load time normal? 
or is there an issue which is slowing down the use of the library? 
what avenues exist to speed up this process? (selective loading?)

some data from the target hardware:

```
# python -c "import can; print(can.__version__)"
3.3.4
```

```
python -X importtime -c 'import can'
import time: self [us] | cumulative | imported package
import time:      8001 |       8001 | _frozen_importlib_external
import time:      2443 |       2443 |   time
import time:      4229 |       6671 | zipimport
import time:      1298 |       1298 |     _codecs
import time:      8647 |       9945 |   codecs
import time:      5928 |       5928 |   encodings.aliases
import time:      9471 |      25343 | encodings
import time:      4001 |       4001 | encodings.utf_8
import time:      1948 |       1948 | _signal
import time:      5758 |       5758 | encodings.latin_1
import time:      1175 |       1175 |     _abc
import time:      5806 |       6980 |   abc
import time:      5761 |      12741 | io
import time:      1500 |       1500 |       _stat
import time:      5071 |       6571 |     stat
import time:     21029 |      21029 |     _collections_abc
import time:      3479 |       3479 |       genericpath
import time:      4619 |       8097 |     posixpath
import time:     11335 |      47031 |   os
import time:      4056 |       4056 |   _sitebuiltins
import time:      4965 |       4965 |   sitecustomize
import time:      1910 |       1910 |   usercustomize
import time:     10528 |      68489 | site
import time:      5153 |       5153 |   __future__
import time:      7336 |       7336 |         types
import time:     18057 |      25393 |       enum
import time:      1575 |       1575 |         _sre
import time:      7989 |       7989 |           sre_constants
import time:      9148 |      17137 |         sre_parse
import time:      8350 |      27061 |       sre_compile
import time:      1911 |       1911 |             _operator
import time:      9692 |      11602 |           operator
import time:      4421 |       4421 |           keyword
import time:      6125 |       6125 |             _heapq
import time:      6023 |      12148 |           heapq
[... 185 lines cut ...]
import time:     10972 |    1862066 |     can.io.logger
import time:      7141 |       7141 |     can.io.player
import time:      7074 |    1876279 |   can.io
import time:     79854 |      79854 |     aenum
import time:      6683 |       6683 |     can.broadcastmanager
import time:     13407 |      99943 |   can.bus
import time:      8590 |       8590 |         wrapt._wrappers
import time:     13930 |      22519 |       wrapt.wrappers
import time:      7162 |       7162 |       wrapt.decorators
import time:      8163 |       8163 |       wrapt.importer
import time:      8280 |      46122 |     wrapt
import time:     19123 |      19123 |             _ctypes
import time:      6669 |       6669 |             ctypes._endian
import time:     23153 |      48945 |           ctypes
import time:      5154 |       5154 |           ctypes.util
import time:      5562 |       5562 |           fcntl
import time:      3932 |       3932 |           can.interfaces.socketcan.constants
import time:      6830 |       6830 |           can.interfaces.socketcan.utils
import time:     14206 |      84626 |         can.interfaces.socketcan.socketcan
import time:      7103 |      91728 |       can.interfaces.socketcan
import time:      6487 |      98215 |     can.interface
import time:      6262 |     150598 |   can.thread_safe_bus
import time:      5260 |       5260 |   can.notifier
import time:     13508 |    3066974 | can
```

christiansandberg, 2020-11-15:
One thing that could be done is to load the logger modules on demand similar to how we only load interfaces when needed. But that could affect the official API.

hardbyte, 2020-11-15:
Hmm that is annoyingly slow but I'm not sure what to do about it. Do you mind uploading the import profile?

If a blank user display is the main concern is there anything preventing you from initializing it first, showing a splash screen/progress screen etc before importing can?

christiansandberg, 2020-11-15:
It seems pkg_resources takes 1,5 s to load alone. You could comment out imports that you don't need to make it load faster. But I don't think we can do it without changing the API completely.

kjngineering, 2020-11-16:
So this is all very new to me, I don't have much experience with python (or any interpreted languages really).

Happy to profile the runtime: I tried to run cProfiler but it hang mid execution - I'm not sure why either because of threading or some other issue. I will have another go later.

Definitely willing to try commenting out some dependencies if you have any recommendations on where to start.

This was an interesting read, but didn't really give me an avenue to cut out anything (particularly pkg_resources).
https://dev.to/methane/how-to-speed-up-python-application-startup-time-nkf

kjngineering, 2020-12-16:
@hardbyte @christiansandberg 

Any suggestions on what I can try to selectively load modules? Happy to do some testing if there is some direction available.

Thank you.

felixdivo, 2021-04-25:
> But I don't think we can do it without changing the API completely.

Yes, it seems to be the case that we can't do much here. Sorry.

felixdivo, 2021-06-05:
I'm closing this for now since I don't see what we should change here.

christiansandberg, 2021-06-06:
Maybe we could use [importlib](https://docs.python.org/3/library/importlib.metadata.html#entry-points) instead. It is for 3.8 and newer only so a fallback to setuptools would probably be needed.

felixdivo, 2022-02-06:
I think this is completely covered by #1234