hardbyte/python-can #941: usb2can doesn't receive all the messages

aramyar, 2020-11-19:
I am writing a simple code to send a message and receive an answer (from a microcontroller). usb2can interface receives the first message but doesn't receive the second message. Then it receives the third message but doesn't receive the fourth message and so on. 

My code:

import can
def send_one(bus, msg):
    try:
        bus.send(msg)
        print(msg)
        print("Message sent on {}".format(bus.channel_info))
    except can.CanError:
        print("Message NOT sent")
def receive_one(bus):
    try:
        msg = bus.recv(1)
        if msg is not None:
            print(msg)
            print("Message received on {}".format(bus.channel_info))
        if msg is None:
            print('Timeout occurred, no message')
    except KeyboardInterrupt:
        pass  # exit normally
if __name__ == "__main__":
    bus = can.interface.Bus(bustype='usb2can', channel='4220D68E', bitrate=20000)
    msg = can.Message(arbitration_id=0, data=[0,0,0,0,0,0,0,0], extended_id=False)
    send_one(bus, msg)
    receive_one(bus)
    bus.shutdown()

felixdivo, 2021-06-05:
Well, your code should only receive a single message and exit. You call `receive_one()` only once.

Does some problem occur with the above code?

felixdivo, 2021-06-29:
If you think there is a bug in the library, please open a new issue or comment on this one.

closed 2021-06-29 by felixdivo