hardbyte/python-can #957: PCAN Interface on macOS

atheler, 2020-12-05:
Hello everyone

I ran into some troubles when I wanted to make a simple SDO query on my Mac. It seems, that there are bogus messages coming back from the Bus. Full disclosure: I am very new to everything CAN so it might well be that I am making some stupid mistake.

```python
with can.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=1000000, fd=False) as bus:
    cob = 0x600 + NODE_ID  # SDOrx + NODE_ID
    sdo_req = b'\x40\x00\x10\x00\x00\x00\x00\x00'  # SDO request for index 0x1000 (Device Type)
    req = can.Message(arbitration_id=cob, data=sdo_req, is_extended_id=False)
    for _ in range(10):
        bus.send(req)
        res = bus.recv()
        print(res)
```

Which results in
```
Timestamp: 19140298519195.671875    ID: 80080000000588    S                DLC:  0
Timestamp: 80501843442118.687500    ID: f280080000000588    X   R            DLC: 89
Timestamp: 25614222983540.769531    ID: 43080000000588    S          EI    DLC:  0
Timestamp: 27303072843804.707031    ID: 80080000000588    S                DLC: 82    01 00 04 05 00 00 00 00
Timestamp: 33776997308149.796875    ID: 43080000000588    S          EI    DLC:  0
Timestamp: 213920982402969.625000    ID: 43080000000588    S          EI    DLC:  0
Timestamp:   102871.084000    ID: 43080000000588    S          EI    DLC:  0
Timestamp: 40532396749205.554688    ID: 43080000000588    S          EI    DLC:  0
Timestamp: 62487444932636.718750    ID: 43080000000588    S          EI    DLC:  0
Timestamp: 99360666881732.656250    ID: 80080000000588    S                DLC:  0
```

I accessed the PCAN with a small C++ program and there it worked fine and I was able to read out the correct device type.

After toying around with the `PCANBasic` class I found, that this seems to be related to the `TPCANMsgMac()` / `TPCANMsg()` structures. I made the following observations:

| Write    | Read   | Resulting Messages |
|----------- |-------------|-------|
|TPCANMsgMac | TPCANMsgMac | Bogus |
|TPCANMsgMac | TPCANMsg    | Works with some SDO aborts |
|TPCANMsg    | TPCANMsgMac | Bogus |
|TPCANMsg    | TPCANMsg    | Works |

This is as far as I got. Am I missing something? Does someone know how this could be fixed?

Thank you!

Best
Alex

Ps: My setup:
macOS 10.15.7 (19H2)
python-can 3.3.4
Python 3.8.6
libPCBUSB.0.9.dylib
PEAK-System PCAN-USB FD

julianpass, 2020-12-14:
Today I had also some strange behaviour as all my python can based scripts were not working after installing libPCBUSB0.9
It was also due to not expected messages and especially because of their wrong identifiers.
I switched back to the older 0.8.1 version of the libPCBUSB which did the trick for me.
Changelog to 0.9 Says that there is some change in the data types which probably could cause the issue.

_- Version 0.9 of 06/25/2020:
  Data types from <MacTypes.h> as replacement for PeakÂ´s usage of Windows BYTE, 
  WORD, DWORD and UINT64 (field CAN ID is now 32-bit wide).
  Synchronized the API to PeakÂ´s PCANBasic version 4.4.0.288
  Parameter PCAN_DEVICE_ID can be read in pre-initialization state.
  Fixed some bugs with status bit: PCAN_ERROR_QXMTFULL, PCAN_ERROR_XMTFULL, 
  PCAN_ERROR_QOVERRUN, PCAN_ERROR_BUSLIGHT.
  Added a delay of 1 usec in CAN_Write to run on macOS 10.15 (Catalina)._

you can download the older versions here: https://www.uv-software.de/dokuwiki/doku.php?id=uvs:programs:pcbusb_library

atheler, 2020-12-17:
I am currently using this small workaround. Thank you @mac-can for the support!

```python
import sys
import re
from ctypes import byref, c_ushort, cdll, create_string_buffer, sizeof

from can.interfaces import pcan

PCAN_NONEBUS = c_ushort(0x00)
PCAN_EXT_SOFTWARE_VERSION = c_ushort(0x86)

def patch_pcan_on_darwin():
    """Temporary workaround for an issue with the PCAN interface on a Mac and
    libPCBUSB version 0.9 (32/64 bit data type issue). Patches PCAN message
    structures in python-can (can.interfaces.pcan).

    Usage:
        >>> import can
        ... import canopen
        ... # Import can & canopen before patching
        ... patch_pcan_on_darwin()
    """
    if sys.platform == 'darwin':
        # Get version number from dylib
        lib = cdll.LoadLibrary('libPCBUSB.dylib')
        buf = create_string_buffer(256)
        res = lib.CAN_GetValue(PCAN_NONEBUS, PCAN_EXT_SOFTWARE_VERSION, byref(buf), sizeof(buf))
        if res:
            raise RuntimeError('Could not get value from lib!')

        ptr = b'.+version (\d+)\.(\d+)\.(\d+)\.\d+'
        m = re.match(ptr, buf.value)
        major, minor, micro = map(int, m.groups())
        if minor >= 9:
            for mod in [pcan.basic, pcan.pcan]:
                mod.TPCANMsgMac = mod.TPCANMsg
                mod.TPCANMsgFDMac = mod.TPCANMsgFD
```

mac-can, 2021-01-14:
@python-can, please apologize any inconvenience due to the 64-bit/32-bit confusion on the ID field. It was a stupid carelessness error at the beginning of the implementation and was fixed with version 0.9 of the PCBUSB library; a CAN message identifier is at most 29-bit wide.

In the meantime I've released version 0.10 and fixed my Python example. I´m using the "original" Python wrapper from PEAK and only had to add the following two lines to make the example run:
```
--- C:\Projekte\CAN\DRV\Drivers\PCANBasic\doc\PCANBasic.py 2021-01-13 22:54:49.000000000 +0100
+++ C:\Users\haumea\OneDrive\Austausch\MacCAN\PCANBasic.py 2021-01-13 23:01:18.000000000 +0100
@@ -377,12 +378,14 @@
    """
    def __init__(self):
        # Loads the PCANBasic.dll
        #
        if platform.system() == 'Windows':
            self.__m_dllBasic = windll.LoadLibrary("PCANBasic")
+       elif platform.system() == 'Darwin':
+           self.__m_dllBasic = cdll.LoadLibrary("libPCBUSB.dylib")
        else:
            self.__m_dllBasic = cdll.LoadLibrary("libpcanbasic.so")
        if self.__m_dllBasic == None:
            print ("Exception: The PCAN-Basic DLL couldn't be loaded!")

    # Initializes a PCAN Channel
```
My suggestion is to undo all macOS specific patches from the python-can code and also use the "original" Python wrapper from PEAK with these two additional lines. Of course, this requires at least version 0.9 of the PCBUSB library. But latest is greatest.

Schöne Grüße aus Berlin.

felixdivo, 2021-04-23:
I assume this is fixed by #976.

closed 2021-04-23 by felixdivo

atheler, 2022-01-19:
Big thanks to everyone for fixing this! I have 4.0.0.dev2 on my machine and it works like a charm.