hardbyte/python-can #958: Cannot Generate Msg Outside Send Command

connoranderson, 2020-12-09:
Noticed an odd behavior on ARM32 architecture, but not on my desktop. When sending a message where the message is generated in the same command as the ".send" command, there is no errror. However, if message is generated and fields are populated there is a socket error.

```
In [1]: import can

In [2]: bus = can.interface.Bus("can0", bustype='socketcan', bitrate=500000)

In [3]: bus.send(can.Message())

In [4]: bus.send(can.Message(arbitration_id=0x80, data=[]))

In [5]: bus.send(can.Message(arbitration_id=0x80, data=[1,2,3]))

In [6]: bus.send(can.Message(arbitration_id=0x80, data=[1,2,3,4]))

In [7]: bus.send(can.Message(arbitration_id=0x80, data=[1,2,3,4,5]))

In [8]: bus.send(can.Message(arbitration_id=0x80, data=[1,2,3,4,5,6]))

In [9]: bus.send(can.Message(arbitration_id=0x80, data=[1,2,3,4,5,6,7]))

In [10]: bus.send(can.Message(arbitration_id=0x80, data=[1,2,3,4,5,6,7,8]))

In [11]: msg = can.Message()

In [12]: msg.arbitration_id=0x80

In [13]: msg.data=[1,2,3,4,5,6,7,8]

In [14]: bus.send(msg)
---------------------------------------------------------------------------
CanError                                  Traceback (most recent call last)
<ipython-input-14-5520df1424f9> in <module>()
----> 1 bus.send(msg)

/usr/local/lib/python2.7/dist-packages/can/interfaces/socketcan/socketcan.pyc in send(self, msg, timeout)
    656                 # Timeout
    657                 break
--> 658             sent = self._send_once(data, msg.channel)
    659             if sent == len(data):
    660                 return

/usr/local/lib/python2.7/dist-packages/can/interfaces/socketcan/socketcan.pyc in _send_once(self, data, channel)
    679                 sent = self.socket.send(data)
    680         except socket.error as exc:
--> 681             raise can.CanError("Failed to transmit: %s" % exc)
    682         return sent
    683 

CanError: Failed to transmit: [Errno 22] Invalid argument
```

```
[volley@bush_1:~]$ candump can0 | grep 000
  can0  00000000   [0] 
  can0  00000080   [0] 
  can0  00000080   [3]  01 02 03
  can0  00000080   [4]  01 02 03 04
  can0  00000080   [5]  01 02 03 04 05
  can0  00000080   [6]  01 02 03 04 05 06
  can0  00000080   [7]  01 02 03 04 05 06 07
  can0  00000080   [8]  01 02 03 04 05 06 07 08
```

hardbyte, 2020-12-10:
:eyes: Interesting! 

I suspect it will work if you construct the message fully using the initializer though. Can you try:

```python
msg = can.Message(arbitration_id=0x80, data=[1,2,3,4,5,6,7,8])
bus.send(msg)
```

If you take the approach of creating an empty `Message` then setting the attributes you will have to set all that depend on what you changed - in this case it looks like the `dlc` won't be correct.

connoranderson, 2020-12-10:
@hardbyte yes that works as well. I tried setting the dlc manually and received the same results, so it's not the dlc field which is causing the issue.

```
In [1]: import can

In [2]: bus = can.interface.Bus("can0", bustype='socketcan', bitrate=500000)

In [3]: msg = can.Message(arbitration_id=0x80, data=[1,2,3,4,5,6,7,8])
   ...: bus.send(msg)
   ...: 

In [4]: msg = can.Message()

In [5]: msg.arbitration_id=0x80

In [6]: msg.data=[1,2,3,4,5,6,7,8]

In [7]: msg.dlc = 8

In [8]: bus.send(msg)
---------------------------------------------------------------------------
CanError                                  Traceback (most recent call last)
<ipython-input-8-5520df1424f9> in <module>()
----> 1 bus.send(msg)

/usr/local/lib/python2.7/dist-packages/can/interfaces/socketcan/socketcan.pyc in send(self, msg, timeout)
    656                 # Timeout
    657                 break
--> 658             sent = self._send_once(data, msg.channel)
    659             if sent == len(data):
    660                 return

/usr/local/lib/python2.7/dist-packages/can/interfaces/socketcan/socketcan.pyc in _send_once(self, data, channel)
    679                 sent = self.socket.send(data)
    680         except socket.error as exc:
--> 681             raise can.CanError("Failed to transmit: %s" % exc)
    682         return sent
    683 

CanError: Failed to transmit: [Errno 22] Invalid argument
```

christiansandberg, 2020-12-11:
Data must be a bytearray.

connoranderson, 2020-12-16:
Correct you are @christiansandberg!

closed 2020-12-16 by connoranderson