hardbyte/python-can #962: How to use the set_filter to get the data

CSG2019, 2020-12-28:
Hi,

I want to send a command to the BUS, then get the response of the special ID and data then do something.
please help to have a look for below code:

``` 
import can

def main():
    bus = can.interface.Bus(bustype='vector', app_name='CANoe', channel=[0])
    pcm_test_msg = can.Message(arbitration_id=0x18DA10F1, dlc=8,
                               data=[0x04, 0x18, 0x00, 0xFF, 0x00, 0x00, 0x00, 0x00])
    bus.send(pcm_test_msg)

    can_filters = [{"can_id": 0x18DAF110, "can_mask": 0x1FFFFFFF, "extended": False}]
    bus = can.Bus(bustype='vector', app_name='CANoe', channel=[0], can_filters=can_filters)

    try:
        while True:
            message = bus.recv(timeout=2)
            if message is None:
                print('no response')
                break
            elif message.arbitration_id == 0x18DAF110:
                print(hex(message.arbitration_id), hex(message.data[0]), hex(message.data[1]), hex(message.data[2]),
                      hex(message.data[3]), hex(message.data[4]))
                break
            else:
                pass
    finally:
        bus.shutdown()

if __name__ == "__main__":
    main()
```

this is from CAN BUS screenshot :
[image]

my question is:
1,I can find the response on CAN, but I can't print the id&data I want to get. could you please help to guide me?
2, I set the timeout=2, after the bus.recv(timeout=2), why didn't the While loop exit.

thank you in advance.

christiansandberg, 2020-12-29:
You create the second bus after you sent the message, so it will not be able to see the message. You have also specified extended to False in your filter but the ID is clearly 29-bit.

CSG2019, 2020-12-29:
Hi christiansandberg,

Thank for the help.
yes, as you said, I created the 2 buses, so it caused I can't get the data I want. After deleted one, it seems to be working well.

about the extended question, yes, my ID should be 29-bit, but when I run below code, I can't print the filter message ,and If I set the extended to False, it can print the filter message. so I don't understand. If you can help to find the reason, please let me know. thank you.
```
def main():
    can_filters = [{"can_id": 0x18DAF110, "can_mask": 0x1FFFFFFF, "extended": True}]
    bus = can.Bus(bustype='vector', app_name='CANoe', channel=[0], can_filters=can_filters)

    for msg in bus:
        print(msg)

if __name__ == "__main__":
    main()
```
the result:

[image]

---------------------------------------------------------------
```
def main():
    can_filters = [{"can_id": 0x18DAF110, "can_mask": 0x1FFFFFFF, "extended": False}]
    bus = can.Bus(bustype='vector', app_name='CANoe', channel=[0], can_filters=can_filters)

    for msg in bus:
        print(msg)

if __name__ == "__main__":
    main()
```
the result:
[image]

felixdivo, 2021-04-14:
Please comment if this is still relevant (as the question is several months old).

closed 2021-04-14 by felixdivo