hardbyte/python-can #963: What are the limitations of CAN FD support?

MaratSabitov, 2020-12-28:
I have a question regarding CAN FD support.

"Limited CAN FD support" declared.
What does it mean,
what are the limitations?

felixdivo, 2021-04-14:
I think this refers to the README (right?). AFAIK, the support for CAN FD is rather complete. Some drivers may not fully support it, but in the general scheme of things this, they should work just fine. Also with the included scripts and file format IO capabilities.

@hardbyte @christiansandberg  Would you approve to removing the "limited" from the README?

hardbyte, 2021-04-14:
@christiansandberg you're the expert so happy to defer to you. I'm inclined to agree with @felixdivo

christiansandberg, 2021-04-15:
I'm definitely no expert but I would say that CAN FD support is about the same status as a lot of other things in python-can. Some implementations are more complete than others and there are certainly aspects of FD that could be implemented if desired. So yes you can remove that disclaimer.

PR #1011 (2021-04-15): New wording for feature list in README
Closes #963.
closed 2021-04-15 by felixdivo