hardbyte/python-can #964: USB2CAN Korlan on Windows

klaasdc, 2020-12-31:
I am trying to use an 8devices Korlan usb2can adapter in Windows 10 64bit. The adapter is using the WinUSB driver and sucessfully captures CAN traffic inside wireshark, so I assume it to be functional.

Then I made a simple test in Python 3.8, to open the bus as instance of Usb2canBus. Initially this did not work because find_serial_devices() inside serial_selector.py tries to match with the legacy usb2can devices that had a serial number starting with "ED".
I modified the code as such:

```
def find_serial_devices(serial_matcher="ED"):
    """
    ...
    """
    objWMIService = win32com.client.Dispatch("WbemScripting.SWbemLocator")
    objSWbemServices = objWMIService.ConnectServer(".", "root\cimv2")
    items = objSWbemServices.ExecQuery("SELECT * FROM Win32_USBControllerDevice")
    ids1 = (item.Dependent for item in items if "VID_0483&PID_1234" in item.Dependent)
    ids = (d.strip('"')[-8:] for d in ids1)
    return [id for id in ids]
```

Now the call to CanalOpen returned 0 until I swapped .dll's from the 8devices website. The tools/doc page at the 8devices wiki is a bit of a mess and very confusing. Some files are for the older usb2can. Eventually a non-zero Canal dll handle was returned with the dll from https://www.8devices.com/media/products/usb2can_korlan/downloads/usb2can_winusb.zip

However, now the code is running without errors, but I do not see any traffic on the bus while sending.
When I put the device in loopback mode and send, python-can receives a message with length of 0.

```
import os
import ctypes
from can.message import Message
import time

if __name__ == '__main__':
    check = ctypes.WinDLL(r'./usb2can.dll')
    from can.interfaces.usb2can.usb2canInterface import Usb2canBus
    
    bus = Usb2canBus(bitrate=500000, flags=0x00000001)#Loopback
#     bus = Usb2canBus(bitrate=500000, flags=0x00000008)
    
    data = bytearray([1,2,3,4,5,6,7,8])
    msg = Message(arbitration_id=1872, is_extended_id=False, data=data, extended_id=False)
    
    while True:
        bus.send(msg)
        print(msg)
        rcv_msg = bus.recv(timeout=1000)
        if rcv_msg is not None:
            print(rcv_msg)
        time.sleep(1)
```
Output of this code is:
```
Timestamp:        0.000000        ID: 0750    S                DLC:  8    01 02 03 04 05 06 07 08
Timestamp:        0.000000        ID: 0000    S                DLC:  0
Timestamp:        0.000000        ID: 0750    S                DLC:  8    01 02 03 04 05 06 07 08
Timestamp:        0.000000        ID: 0000    S                DLC:  0
```

At this point I'm out of ideas on how to get the Korlan usb2can to work on Windows, but I'm willing to try any suggestions :)

tomcunn, 2021-11-10:
I messed with similar and have about given up hope. 

Python 3.10
pywin32=302
python-can 2.2.1
Used the dll labeled WinUSB CANAL DLL v1.0.0 (32/64 bit)

Used similar code, can receive without issue, but no transmits. After updating to 3.10, had to make some changes to code for locating .dlls and for the bit rate because there was some issue there. 

I bought the Korlan thinking it would be a quick solution using python-can, I like the look of the device, but need to be able to transmit in python.

klaasdc, 2021-11-10:
I received some code from 8devices but I never got it to work on my 64 bit system. Even their test application fails to start or crashes. At some point it was "working" without errors in Python but in reality never sent out or received anything. I gave up and moved on to another USB-CAN adapter that works both in Win and Linux.