hardbyte/python-can #965: BLF reader is missing direction of CAN FD Message

fabiocrestani, 2021-01-08:
When parsing a CAN_FD_MESSAGE_64, the direction byte is ignored:
https://github.com/hardbyte/python-can/blob/7ddea9e17f26425fbe943e3151e0e47baf0d5a22/can/io/blf.py#L315

closed 2021-06-08 by mergify[bot]

Matheus-Garbelini, 2022-05-17:
Currently, direction of CAN messages are only parsed for CAN_FD_MESSAGE_64. Is there any way we can also save BLF with the direction information for CAN FD or normal CAN frames?
Is there some new format on the step that allows something like "CAN_MESSAGE_64" so we can use direction for classical CAN messages?

felixdivo, 2022-05-17:
(I'm out, sorry.)

felixdivo, 2022-05-18:
@Matheus-Garbelini Yes. That's right. Just so that you don't wait for my feedback on this.

fabiocrestani, 2022-05-31:
@Matheus-Garbelini By inspecting the code it seems that all CAN messages have a `is_rx` field that you can use to get the direction. Maybe you could you provide a sample file for further investigation?

Matheus-Garbelini, 2022-06-01:
Hi @fabiocrestani, thanks for reply.
Unfortunatelly, the main point of my concern is that we cannot save the BLF with direction information.
Readinf is fine trough the `is_rx` flag as you suggested, but informatino is lost if the user tries to save the same capture via BLF Writer.

PR #966 (2021-06-08): Add msg direction for CAN-FD in BlfReader
Fixes #965