hardbyte/python-can #967: viewer.py with options to publish scaled & labeled CAN signals to MQTT

dalklein, 2021-01-12:
Hello,   thanks for this nice working python CAN code!
     I modified the  python-can/can/viewer.py  (renamed to viewmqtt.py),  to listen on CAN and publish signals out to MQTT.
Any interest to include it?    I'm not sure my changes are up to any standard, may be a little hacky, but it works as before unless the different command line option is given and labels are added to the signal decode parameter file.

If there's any interest, please let me know and I'll do a pr after any further improvements needed.  Or say if there's anything recommended to update or change to make it acceptable.  Or if there's some existing tool for the purpose, that I didn't see.

high level changes / notes
imports paho-mqtt,  and publishes to mqtt on localhost:1883    (needs mosquitto or broker running locally if -q is used)
added arg parsing to label signals (assuming they have been filtered & scaled, then see if labels are there for them
-q  option for 'quiet',  enables publishing to mqtt, and stops use of stdscr, so it can be run in the background.
I did not do anything yet to downsample potentially high rate CAN signals appropriately for MQTT.  This could be needed for signals at higher rate than .5 or .1 sec, to avoid too many MQTT messages.

Changes are here, based on last release I think, on branch can2mqtt:     https://github.com/dalklein/python-can/tree/can2mqtt                 new file  can/viewmqtt.py,   and new example decode file si-decode_label.txt

Thanks,   Dale

`Count   Time           dt          ID          DLC  Data                     Parsed values
153     15.250335      0.099617    0x001       2    00 00                    0.000
153     15.246226      0.099594    0x002       2    00 00                    0.000
1533    15.319339      0.009096    0x010       8    03 03 E7 3B D3 56 0F 0A  771 15335 22227 25755
153     15.250459      0.099571    0x020       8    00 00 00 00 00 00 00 00  0 0 0 0
1532    15.314229      0.010370    0x031       8    0B 24 31 D7 DB FF 0A 00  9227 -10447 -37 10
153     15.244714      0.099604    0x041       8    DA FF 5D 03 1C 00 E5 D6  -38 861 28 -10523
153     15.244836      0.099606    0x051       8    00 00 00 00 00 00 00 00  0 0 0 0
15      14.388234      0.999473    0x060       2    02 38                    56.800 BatTgtV
1533    15.320720      0.010386    0x070       8    00 00 00 00 00 00 00 00  0 0 0 0
28      15.284337      0.554245    0x181       8    C4 D6 01 00 3E D7 01 00  120.516 InvV 120.638 GrdV
8       14.343360      2.038847    0x191       8    42 4B 00 00 68 1A 00 00  19266 0 6760 0
28      15.174349      0.554373    0x1B1       8    00 00 00 00 00 00 00 00  0 0 0 0 0 0 0 0
27      14.839966      0.551357    0x1C1       8    FE E9 00 00 EB 00 00 00  59.902 InvHz 0 b34 58.750 InvAbsW 0 b78
`

felixdivo, 2021-04-14:
This looks like a neat tool. However IMHO, this does not quite fit into the scope of this library.

I think it would be useful to have a "other nice CAN-related (Python) tools". Your script would fit there.

felixdivo, 2021-05-05:
I added this idea to #1041. It will be tracked there.

closed 2021-05-05 by felixdivo