hardbyte/python-can #970: How can I change the data of a message before it send periodically?

Dongeeee, 2021-01-16:
I want to send a message 0x30C every 100ms, every time before the message sent to bus , one byte of the data should increase 1.  

How can I implement this? I think changing the data every 100ms is not so good, when the bus load is too high, message 0x30C period is not 100ms exactly, which cause the byte not increase 1 or will increase 2.

Can I change the message data(or callback a function) before message sent to bus automatically(_**send_periodic()**_)?

Appreciate for your reply.

hardbyte, 2021-01-19:
Something like [this example](https://github.com/hardbyte/python-can/blob/develop/examples/cyclic.py#L56-L82)?

Dongeeee, 2021-01-19:
@hardbyte Hi, thanks for your support. But when I refer to your method, something different happens.
Function safety related message usually update like this：
I want update cyclic message before every time it sent, so that rolling counter is continuously. 
And your example seems update message Occasionally. I have to call modify_data() every 200ms, when high CAN bus load, real CAN message period may longer than 200ms, which may result rolling counter not continuously(repeat or skip). CAN I call a function to modify data every time message sent by send_periodic?

time | ID | byte0 | byte1 | byte2 | byte3 | byte4 | byte5 | byte6 | byte7
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 0x30C | CRC     =sum(B1:B7) | random     update | random     update | random     update | random     update | random     update | random     update | rolling   counter     Initial value 0
0.2 | 0x30C | CRC     =sum(B1:B7) | … | … | … | … | … | … | 1
0.4 | 0x30C | CRC     =sum(B1:B7) | … | … | … | … | … | … | 2
0.6 | 0x30C | CRC     =sum(B1:B7) | … | … | … | … | … | … | 3
… |   |   |   |   |   |   |   |   | …
3.0 | 0x30C | CRC     =sum(B1:B7) | … | … | … | … | … | … | 0x0F
3.2 | 0x30C | CRC     =sum(B1:B7) | … | … | … | … | … | … | 0
… |   |   |   |   |   |   |   |   | …

hardbyte, 2021-01-19:
Oh I see, no we don't have anything currently implement as part of the core library. It is tricky because where possible we offload the periodic sending to the lower level interfaces. 

Have a look at one proposed solution in https://github.com/hardbyte/python-can/pull/703

Of course there is nothing preventing implementing it yourself in user code, just the timing will only be as good as Python can manage.

Dongeeee, 2021-01-20:
> Oh I see, no we don't have anything currently implement as part of the core library. It is tricky because where possible we offload the periodic sending to the lower level interfaces.
>
> Have a look [...]

Thanks for your  support. I'll try it according your link. Thank you very much.

felixdivo, 2021-04-14:
@Dongeeee Did this solve your problem?

This is tracked in #703 and #809.

Dongeeee, 2021-04-14:
> @Dongeeee Did this solve your problem?
> 
> This is tracked in #703 and #809.

Not yet. I'll handling something else recently.

felixdivo, 2021-04-14:
OK. Since it is tracked elsewhere, I'll close it here. Feel free to comment if the other approaches do not fit your use case and we will discuss it further.

closed 2021-04-14 by felixdivo

felixdivo, 2023-03-28:
This issue is closed. See the referenced ones above.