hardbyte/python-can #973: blf read header information to get start date of measurement

nexulm, 2021-01-27:
Did I miss something? 
How can I extract the header lines from a .blf files to get the date of measurement start?
.asc starts always with this first line, i.e:
date Fre Jan 22 13:20:12.005 2021

I need the information of Date and time of measurement start.
start_timestamp returns only "13:12:12.005" and not the date (here: 2021/01/22)

christiansandberg, 2021-01-27:
The start_timestamp is in seconds since the Epoch (Unix timestamp). What is the raw float value?

nexulm, 2021-01-27:
Good hint, didn't realize this. :-(

Code is up and running now. ;-)
!!!
    fCANin      = can.io.BLFReader(infile)
    # read measurement start_timestamp from .blf file
    dt_mtstart  = format(datetime.fromtimestamp(fCANin.start_timestamp), '%Y-%m-%dT%H:%M:%S.%f')
!!!

closed 2021-01-27 by nexulm