hardbyte/python-can #974: virtual.rst example doesn't work

NiallDarwin, 2021-02-01:
Copying the example code from virtual.rst (or in the docs here: https://python-can.readthedocs.io/en/master/interfaces/virtual.html) and then running it gives an error on both Spyder and Python2 on a Pi. 
I'm an absolute novice on coding, Python & CAN so its probable I've got something wrong. Its disconcerting when examples don't work!

NiallDarwin, 2021-02-01:
I have created a working version and (I think) submitted it to be integrated. Let me know if I've done it right or if you don't see it.

hardbyte, 2021-02-01:
Hi @NiallDarwin it sounds like you might be using a version of Python no longer supported by python-can. Have a look at the table on our readme:

[image]

If you need more help can you tell us what version of Python and what version of python-can you're using?

NiallDarwin, 2021-02-01:
Hi, thanks for the response. My versions are:
On Pi:
Python 2.7.16
Can library 3.3.4

On Spyder 4.1.3 (?) via Anaconda on Windows10:
Python 3.7.7
Can library 3.1.0

Both look like they should be compatible by your list.

hardbyte, 2021-02-01:
Hmm okay sounds like a legit bug, thanks for opening the issue. In the future its a good idea to try include the output that you're seeing, and what you were expecting. For instance if there is an exception that might give us a clue as to what is going on.

You say you've submitted a working version but I don't see any [pull requests](https://github.com/hardbyte/python-can/pulls) from you? There is some good [documentation from github](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) about how to go about that.

NiallDarwin, 2021-02-01:
OK, points noted. 

>You say you've submitted a working version but I don't see any pull requests from you? There is some good documentation from github about how to go about that.

I'm not sure what I did the first time but I've now submitted a pull request.

>In the future its a good idea to try include the output that you're seeing, and what you were expecting. For instance if there is an exception that might give us a clue as to what is going on.

I was expecting the code to run without outputting anything. What I got was:

**Spyder Output**
runfile('C:/Users/NiallX1/.spyder-py3/untitled0.py', wdir='C:/Users/NiallX1/.spyder-py3')
Traceback (most recent call last):

  File "C:\Users\NiallX1\.spyder-py3\untitled0.py", line 18, in <module>
    assert msg1 == msg2

AssertionError

---------------------------------

Thank you so much for creating and maintaining this code. It is greatly appreciated.

hardbyte, 2021-02-02:
Oh, I see you've created a pull request in your own repository - https://github.com/NiallDarwin/python-can/pull/1

What you probably meant to do is to create a PR against my "develop" branch https://github.com/hardbyte/python-can/compare/develop...NiallDarwin:NiallDarwin-virtualExamplePatch1
 then I or one of the many maintainers/contributors to python-can will review your patch and integrate it.

hardbyte, 2021-02-02:
Happy to do it myself, but I like to encourage contributions :-P

hardbyte, 2021-04-04:
Closed by #975

closed 2021-04-04 by hardbyte